e2nIEE/pandapower #789: time series simulation with p_mw and q_mvar per bus and time step?

chrisnexxtlab, 2020-05-27:
I have a question related to time series simulation.

I am building controllers for loads, sgen and storage like this:

```
for n in range(0, len(pod_array)):
     pod = pod_array[n:n+1].pop() 
     const_load = pandapower.control.ConstControl(net, element=‘load', element_index=net.load.index, variable='p_mw',  data_source=ds, profile_name=pod) 
```
So far so good. But I don’t want to simulate a network having purely resistive loads. How can I add “_q_mvar_” values for the same network per bus and time step?

Is that a functionality that is not yet provided with ts simulation by pp 2.2.2 or did I miss something essential?

Thank you very much in advance!

FlorianShepherd, 2020-05-28:
You just have to add a second const_controller, wie a second DataFrame including the `q_mvar` values:

```
for n in range(0, len(pod_array)):
     pod = pod_array[n:n+1].pop() 
     const_load = pandapower.control.ConstControl(net, element=‘load', element_index=net.load.index, variable='p_mw',  data_source=ds, profile_name=pod) 
     const_load = pandapower.control.ConstControl(net, element=‘load', element_index=net.load.index, variable='q_mvar',  data_source=ds_qmvar, profile_name=pod) 
```

chrisnexxtlab, 2020-05-28:
Thank you @FlorianShepherd. Indeed straight forward, working nicely.

FlorianShepherd, 2020-05-29:
Great :+1:

closed 2020-05-29 by FlorianShepherd