e2nIEE/pandapower #790: changing attribute initial_powerflow in controller

SimonRubenDrauz, 2020-05-28:
As pandapipes is based on pandapower I would like to rename the control attribute 'initial_powerflow ' to 'intial_run', in order to prevent deletion of this attribute for each pandapipes controller and change it to initial_pipeflow. Would this be possible?

A solution has been pushed on branch 'initial_powerflow' found in my fork!

FlorianShepherd, 2020-06-02:
I don't mind changing the name as long as the description is fine. I think it's important to know what it does (run an additional "time consuming" power flow  / pipeflow before each time step). Would you create a PR  @SimonRubenDrauz ?

SimonRubenDrauz, 2020-08-25:
done. Changes have been implemented!

closed 2020-08-25 by SimonRubenDrauz