e2nIEE/pandapower #793: rst file line endings in git

jurasofish, 2020-05-30:
Does anyone have the issue where git reports rst files are modified due to line endings?
Looks like this in github desktop (convert_format is a legit change).
If I remove ``*.rst text eol=lf`` from ``.gitattributes`` it's fine. I'm on MacOS.

[image]

FlorianShepherd, 2020-06-02:
I haven't experienced this on Linux so far. Nonetheless, I think it's a good idea to integrate that since it might be a problem for other users as well. Thanks @jurasofish

jurasofish, 2020-06-02:
hopefully it doesn't mess things up for anyone :)

FlorianShepherd, 2020-06-02:
I hope so as well. Since it is only a possible problem for the *.rst files, I think we could give it a try. 

Here is some more information on the topic: https://help.github.com/en/github/using-git/configuring-git-to-handle-line-endings

With `eol=lf` one forces it to the Unix version as far as I get it

jhmenke, 2020-06-02:
can you verify first that you have set the git autocrlf to input?

git config --global core.autocrlf

(set with *git config --global core.autocrlf input*)

FlorianShepherd, 2020-06-02:
Ok, apparently this has some history. @lthurner added the lf flag three months ago. Maybe he can clarify why before we merge the request.

lthurner, 2020-06-03:
I think @SimonRubenDrauz  can say something about that?

SimonRubenDrauz, 2020-06-03:
Yes, I can comment on this. I recommended that change due to some problems with pypi. I had some troubles with uploading the .zip-package to pypi as my README.rst-file and CONTRIBUTING.rst-file was in Windows/Dos, not in UNIX Format. It took me really long to find out it was an issue of the format. In order to prevent that some other user has the same problem again, I recommended to add *.rst text eol=lf. What does it: All .rst files which are synced locally are forced to be UNIX instead of Windows/Dos format. By this, I can make sure that they are always in UNIX format and no one will have the same problem as I had.  
What do I recommend now. I think there are two solutions:
1.) In your case, these files were initially saved as Windowd/dos (before lthurner made the change) and pushed. Once you pulled it, they changed automatically to UNIX. That's the reason why the change of the format endings was detected (don't ask me why no one else had a similiar problem before). You could push that change, make a pull request and change the format of all .rst-filles unifromaly to UNIX.
2.) Instead of forcing all files to be UNIX we could just make the change for the pypi-relevant files. But I wouldn't recommend that because I would say it is best if we all work with the same line endings no matter if we are on Windows, Linux or Mac, to prevent similiar problems as I had once.
What you think?

jurasofish, 2020-06-03:
@jhmenke I'd played around with those settings with no success, I think it might be overridden by the gitattributes file anyway.

Thanks for the info @SimonRubenDrauz! I think what I'll do is just commit the line ending changes, seems like that's something that's inevitably going to have to happen anyway. Also looks like git wanting to change the line endings is intermittent for me as well... I'll just commit them as they come up.

closed 2020-06-03 by jurasofish