e2nIEE/pandapower #798: Strange Voltage rise at the beginning of the Voltage Profilep

na1203132, 2020-06-02:
Hi there,

I have an issue with the voltage profile plotting. I have tested the function on many networks, but it gives the same issue. The issue is that there is a strange voltage rise over the slack bus. This can be observed from the voltage profile, there is a spike at the beginning of the profile and then the normal shape of the voltage drop continues. I am wondering why is this happening? What causes this voltage rise at the slack substation ?  

For your reference, to visualize what I mean, try to run the following code snippet: 

```
import pandapower as pp
from pandapower.plotting.plotly import simple_plotly
from pandapower.networks import mv_oberrhein
import matplotlib.pyplot as plt
net = mv_oberrhein()
pp.plotting.plot_voltage_profile(net)
plt.show()
```

Thank you in advance.

lthurner, 2020-06-03:
That is the voltage drop over the transformer :)

na1203132, 2020-06-03:
> That is the voltage drop over the transformer :)

But, as you said, it is a voltage drop, not voltage rise! What I am seeing is a spike, but not an actual drop. For example, I could run the same network on DigSilent, I don't see such spike, but only the normal voltage drop shape. Or maybe I missing out something here. Is this voltage rise related somehow to any built-in automatic tap-changer activated inside the function of the transformer?

lthurner, 2020-06-03:
Yes there is a tap changer that causes this voltage difference (its actually a rise, not a drop in the default settings). Check out the documentation about the transformer here: https://pandapower.readthedocs.io/en/v2.2.2/elements/trafo.html#input-parameters

> For example, I could run the same network on DigSilent, I don't see such spike, but only the normal voltage drop shape. 

If you would model the same transformer with the same tap changer position, you would see the exact same voltage rise/drop. That is thouroghly tested.

na1203132, 2020-06-03:
Thank you very much! I didn't know that it was activated by default. Now I understand. I really appreciate your clarification. 

> Yes there is a tap changer that causes this voltage difference (its actually a rise, not a drop in the default settings). Check out the documentation about the transformer here: https://pandapower.readthedocs.io/en/v2.2.2/elements/trafo.html#input-parameters
> If you have a look at the parameters in the net.trafo table, you will see the tap changer definition. If you change net.trafo.tap_pos, you will see how the voltage profile changes.
> 
> For example, I could run the same network on DigSilent, I don't see such spike, but only the normal voltage drop shape.
> 
> If you would model the same transformer with the same tap changer parameters and position, you would see the exact same voltage rise/drop. That is thouroghly tested.

lthurner, 2020-06-03:
You are welcome. by the way if you create a network with the create functions, the tap changers will default to medium position and you will not see this voltage drop, unless you explicitely set the tap changer. But in this predefined network, there is already a tap changer position set that causes this voltage rise.

closed 2020-06-03 by lthurner