e2nIEE/pandapower #813: Voltage deviations at Oberrhein Network after a disturbance ...

Houshyar20, 2020-06-12:
Hi everybody,

I'm fresh in the Pandapower and electric power system. I have a problem when I want to check the power flow on the MV-Oberrhein network. I will be happy if somebody, particularly Pandapower developers, could help me.
I impose a disturbance on the (active) power static generator at a specific bus, e.g. bus 33, and I want to know how voltage values are changed. I use the attached script (DV_Oberrhein.txt) to check the deviation of voltage magnitude at all buses.
Strangely, I obtain zero values for voltage magnitude deviations at all buses!
Why do I get such a result? I expect, after a disturbance at the (active) power of a static generator, that I obtain different values for some quantities like voltage magnitude at every bus!
I imposed disturbances at all static generators but the results were the same.

Thanks for your reply in advance.
Houshyar

[DV_Oberrhein.txt](https://github.com/e2nIEE/pandapower/files/4772071/DV_Oberrhein.txt)

ascheidl, 2020-06-12:
hi,

for some unknown reason, the scaling properties of the sgens are all 0 in mv_oberreihn.

set it to something more reasonable.

cheers,
alex

Houshyar20, 2020-06-12:
Hi Alexander,

Thanks for your reply. 
I did not understand very well, did you mean my script should be improved by new indexing for sgen to get correct result or I should be wait for probable improve in the pandapower for this network?

Thanks for your reply in advance.

Cheers,
Houshyar

Houshyar20, 2020-06-13:
> Hi Alexander,
> 
> Thanks for your reply.
> I did not understand very well, did you mean my script should be improved by new indexing for sgen to get correct result or I should be wait for probable improve in the pandapower for this network?
> 
> Thanks for your reply in advance.
> 
> Cheers,
> Houshyar

Just a little bit more explanation about what I did:
Actually, as you can see in my short script, I just used the well-known command as follows:

gt=net.sgen.index[net.sgen['bus']==33].tolist()
disturb=net.sgen.p_mw[gt[0]]*20/100
net.sgen.p_mw[gt[0]]=net.sgen.p_mw[gt[0]] + disturb

And I checked, it gives a correct sgen index for mv_oberrhein network, I mean the corresponding net.sgen.index is 10 for the bus 33 and it seems to be reasonable. There is no zero value for this sgen index and also for other properties of sgen. 

Thanks for your clarify in advance.
@ascheidl

closed 2020-06-14 by Houshyar20