e2nIEE/pandapower #814: Short-Circuit: Phase of current

bergkvist, 2020-06-12:
#### It seems like only current in absolute value is returned from the short circuit calculations. 
How can I get the phase of this current as well?

<details><summary>Click to see an example</summary>

```py
import pandapower.shortcircuit as sc
import pandapower.networks as nw

net = nw.create_cigre_network_mv()
net.ext_grid["s_sc_min_mva"] = 100
net.ext_grid["rx_min"] = 0.1

net.line["endtemp_degree"] = 20
sc.calc_sc(net, case="min")
display(net.res_bus_sc)
```
[image]

</details>

## Motivation
I want to select a subnet of my network by using a Thevenin equivalent simplification. 
 - http://hyperphysics.phy-astr.gsu.edu/hbase/electric/acthev.html

### Finding Thevenin equivalent voltage (v_th)
Finding v_th should be easy, I just disable the subgrid I plan on selecting, and measure the voltage at the connection point after running a powerflow.

### Finding Thevenin equivalent impedance (z_th)
To find the z_th I can look at the short circuit current at the connection point. Let's call this current i_sc. Then we have that
```py
# Notice that in order to find the correct thevenin impedance, 
# I need to know the phase difference of v_th and i_sc
z_th = v_th / i_sc
```

I might be off track somewhere, or have overlooked an existing feature for achieving this.

lthurner, 2020-06-15:
The short circuit current is calculated from the impedance, so calculating the short-circuit first and then extracting the impedance from that is a little backwards. There is no isolated function yet that calculates the impedance, but you can debug into the calc_sc function. The impedance matrix is calculated in there somewhere. Either you can isolate that as a function that can be called independently of the short-circuit calculation, or save it during the calculation and access it afterwards.

bergkvist, 2020-06-30:
So if I understand this correctly, the impedance matrix contains z_th for every bus already?

Is there a way of calculating the "momentary short-circuit current" rather than min/max? Or even the momentary z_th for every bus? This would allow for calculating power efficiency for every load (assuming load impedance z_L is known).

Based on the [Maximum Power Transfer Theorem](https://en.wikipedia.org/wiki/Maximum_power_transfer_theorem), it would also be possible to calculate how much "available power capacity" a load has before causing powerflow divergence. (which would be really cool)

lthurner, 2020-08-10:
> So if I understand this correctly, the impedance matrix contains z_th for every bus already?

Not 100% sure we are talking about the same thing, but yes, I would think so.

> Is there a way of calculating the "momentary short-circuit current" rather than min/max? Or even the momentary z_th for every bus? This would allow for calculating power efficiency for every load (assuming load impedance z_L is known).

What does "momentary" mean in this context?

bergkvist, 2020-08-10:
I think "momentary" was a misunderstanding from my side of what "min" and "max" actually means in the pandapower short circuit calculation (`pandapower.shortcircuit.calc_sc`).

Assumptions:
 - `net.load.p_mw`, `net.load.q_mvar` and `net.switch.closed` are the only inputs that change over time. (for a time series scenario)
 - A "momentary" short-circuit calculation computes the short-circuit currents for a specific set of `net.load.p_mw`, `net.load.q_mvar` and `net.switch.closed`.

My understanding right now is that min and max is entirely related to how much power the ext_grid is able to inject into the grid. And that it is entirely unrelated to the load p_mw or q_mvar values.

---

When running a regular powerflow (`pp.runpp`) - you essentially solve for/find the "momentary" load impedances in the network. If you change `net.load.p_mw` for a specific load in the network - all other loads in the network might need to adjust their impedance (in the solution) to retain the same power output.

Setting the impedance for one of these loads to zero (ignoring its p_mw, q_mvar), and then running a regular powerflow would correspond to a short circuit calculation for that load. The actual current going through this load would depend on p_mw for other loads in the network.

lthurner, 2020-09-01:
We implement the short-circuit according to IEC 60909, which says that loads should be ignored entirely in the short-circuit calculation. So load impedances aren't considered at all right now.  The load impedances themselves are so small compared to the short-circuited line that they have virtually no impact, which is why they are usually ignored.

The short-circuit current only depends on the network impedances (lines, transformers) and the assumed voltage at the time of the short-circuit. The assumed voltage can either be an upper worst-case (the highest voltage that could possibly ocurr -> "max") or a lower voltage worst case (the lowest voltage that could possibly ocurr --> "min"). The standard also provides the possibility to calculate a short-circuit with a voltage profile from a power flow. That seems to be closest to what you have in mind. That is not currently implemented, would however be an interesting feature.

SteffenMeinecke, 2021-11-10:
close due to inactivity and no obvious near future TODOs

closed 2021-11-10 by SteffenMeinecke