e2nIEE/pandapower #816: Powermodels test failing locally - Different results

friederikemeier, 2020-06-14:
Hi,

The test_powermodels.py is failing locally. The calculation results are not the same. I don't know, which packages may cause this.
I am using pyjulia 0.5.3 with julia 1.1.1.

My Julia setup:

  [9961bab8] Cbc v0.7.0
  [31a5f54b] Debugger v0.6.4
  [b6b21f68] Ipopt v0.6.2
  [682c06a0] JSON v0.21.0
  [4076af6c] JuMP v0.21.2
  [2ddba703] Juniper v0.6.3
  [c36e90e8] PowerModels v0.17.1
  [438e738f] PyCall v1.91.4

This is the error:

`runfile('C:/Users/fmeier/pandapower/pandapower/test/opf/test_powermodels.py', wdir='C:/Users/fmeier/pandapower/pandapower/test/opf')
PROJ: proj_create_from_database: Cannot find proj.db
Traceback (most recent call last):

  File "<ipython-input-1-df71387ec0de>", line 1, in <module>
    runfile('C:/Users/fmeier/pandapower/pandapower/test/opf/test_powermodels.py', wdir='C:/Users/fmeier/pandapower/pandapower/test/opf')

  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/fmeier/pandapower/pandapower/test/opf/test_powermodels.py", line 558, in <module>
    test_pwl()

  File "C:/Users/fmeier/pandapower/pandapower/test/opf/test_powermodels.py", line 163, in test_pwl
    assert np.isclose(net.res_cost, net.res_gen.p_mw.at[g1])

AssertionError`

These are the values that are compared by the isclose():

`ipdb> net.res_cost
1.001866180629729

ipdb> net.res_gen.p_mw.at[g1]
1.0018863365379396`

The difference ist small, but it's there.

Suggestions & Questions:

- How can we test this better?

- Should we test the json inputs for the run_powermodels.jl ?

- Should we agree on a specific version of powermmodels, julia and so on?

- Which packages are causing this?

- What configurations are you using?

FlorianShepherd, 2020-06-15:
Hi,

Here's my setup:
```
  "Atom"        => v"0.12.10"
  "KNITRO"      => v"0.9.1"
  "JuMP"        => v"0.21.2"
  "CPLEX"       => v"0.6.5"
  "Juno"        => v"0.8.1"
  "Juniper"     => v"0.6.3"
  "Ipopt"       => v"0.6.1"
  "PyCall"      => v"1.91.4"
  "JSON"        => v"0.21.0"
  "PowerModels" => v"0.17.0"
  "SCIP"        => v"0.9.5"
  "Cbc"         => v"0.6.7"
  "Gurobi"      => v"0.8.0"
  "Alpine"      => v"0.1.3"
julia v.1.1.0
pyjulia 0.5.3
```

I get the same result. I think this is just a neglectable solving tolerance issue. Feel free to reduce the numpy isclose atol / rtol values.

lthurner, 2020-06-15:
Regarding the last point:

> Should we set up travis for this? The powermodels tests are skipped on travis at the moment, this is why I can't figure out what the right behaviour would be. -> Yeah good idea. Feel free to do it. I'm not sure how to integrate julia.

I tried to do this when I implemented the original interface. pyjulia has a setup that runs Python and Julia together in travis CI: https://travis-ci.org/JuliaPy/pyjulia. It should be able to look into their travis configuration and set this up for pandapower in a similar way: https://github.com/JuliaPy/pyjulia/blob/master/.travis.yml. I didn't have enough time to do it unfortunately, but feel free to look into it.

friederikemeier, 2020-06-15:
I think that must have something to do with the different version of Ipopt (0.6.2)... That may also be the reason, that the tutorials don't run locally for me. I was fixing the tutorial for the newest version of the ipopt package and now it possibly won't run with your version (0.6.1). So I feel like fixed versions of the julia packages, that are valid for tests and tutorials should be set.

lthurner, 2020-06-15:
>  So I feel like fixed versions of the julia packages, that are valid for tests and tutorials should be set.

I agree. Ideally, we would do this through setting a version in Travis that is tested. So its all down to getting Travis to run with julia.

shimamary, 2020-06-16:
To use Julia in python please consider these points:

1. Don't use Conda: https://pyjulia.readthedocs.io/en/latest/troubleshooting.html, 
2. Don't use Juliapro
3. Delete all your packages and install them again, please try:

- import Pkg
- Pkg.rm("PACKAGE NAME")
- Pkg.add("PACKAGE NAME")
- Pkg.update()
- Pkg.resolve()

I hope these points solve your problem.

friederikemeier, 2020-12-02:
Thanks to Travis we don't have the discrepancy anymore. closed 😄

closed 2020-12-02 by friederikemeier