e2nIEE/pandapower #820: PowerModels: [Win127 Error] if used in combination with conda

lschmelting, 2020-06-17:
**Goal if this issue:**
1. Test a combination of  [image] and debugging the error described below to be able to use conda with PowerModels again.
2. Update the documentation for installation + usage of PowerModels.

**Description:**
In October 2019, i was able to use conda + PowerModels in pandapower with Spyder. Yesterday @SteffenMeinecke noticed he cannot use it as intended:
```
from julia.api import Julia
jl= Julia(compiled_modules=False)
```
works in an anaconda prompt, but fails if executed in spyder ([WinError 127]).  If PowerModels is called via pandapower in an anaconda prompt, the same error occurs (Both with the current code in pandapower/pandapower/opf/run_powermodels.py or by replacing it with the lines above to circumvent the [image].

Also note that conda is not officially supported by both [image] and [image], so we are on our own if issues appear at any time.

lschmelting, 2020-06-17:
Any ideas for improvement for this pull request? #821

lschmelting, 2020-06-17:
I was able to call PowerModels with the modified Code. It's kind of strange that i had to move `from julia import Main` outside of the try-block to make this work. I also modified the Code so users can use their own sysimage_file (compiled_modules=False is VERY slow)

rbolgaryn, 2020-10-13:
Hi @lschmelting ,

can the issue be closed?

lschmelting, 2020-10-14:
The issue can be closed, the PR must be updated before merging.

closed 2020-10-14 by lschmelting