e2nIEE/pandapower #827: OPF line loading limits - current or power?

friederikemeier, 2020-06-30:
Hey!

In pandapower, the line loading is defined in percent of the actual current of the maximum current net.line.max_i_ka.
This is why the standard call of the pypower OPF uses the matpower/pypower parameter opf_flow_lim="I" per default.

In PowerModels, the standard formulation does not constrain the current, but the apparent power. What our interface now does, is to calculate the apparent power at 1.0. p.u. voltage and enter this as the maximum apparent power of the line.

It is evident, that this leads to an inconsistency, because we can never hit the same results with the pypower and PowerModels OPF. I suggest to extend both interfaces, pypower pp.runopp and PowerModels OPF pp.runpm_ac_opf by the following parameter:

opf_flow_lim (that's what the parameter is called in pypower) which can have the values "I" and "S", so the user can decide which formulation he wants.
The default value should be "I" in my opinion.

What do you think?

FlorianShepherd, 2020-07-08:
Sounds good to me. I think it must be added in `build_branch` somewhere.

friederikemeier, 2020-10-09:
So... I am done with the implementation... I am about to make a pull request..Should I change the default of line loading constraints of the Powermodels interface to "I" now?  @lthurner? @Inga ? @Maryam ?

friederikemeier, 2020-10-14:
It still defaults to "S" now, I don't want to change the behaviour just like this... maybe we can change the defaults after I implemented and tested the current limits also for TNEP. But I am closing this for now.

closed 2020-10-14 by friederikemeier

rbolgaryn, 2020-10-19:
I think it is reasonable to use current as loading limit and for the calculation of loading