e2nIEE/pandapower #828: line capacitance: line-to-earch, line-to-line or total operating capacity?

marhofmann, 2020-06-30:
Hello pandapower team,

regarding to the question in the title, i wonder which type of line capacity the line types need to have. The documentation on lines calls it "line capacitance in nano Farad per km" which is not distinct to me but i assume the total operating capacity is meant.
So if i want to calculate the total line-to-earth capacity of my entire gird its not just the sum of all lines within the gird, rigtht?
Additionally i suggest to make the documentation a bit more clear.

Thanks in advance,
Martin

bergkvist, 2020-06-30:
My understanding is that line capacitance describes internal capacitance between the conductors of a line/cable (net.line). If you only have a single conductor, this value would be 0.

[image]

In general you will see that cables (underground/underwater) have higher capacitance than lines (above ground), because the conductors are closer together.

## High line capacitance can cause powerflow divergence or increase losses. 

Using 0 is an optimistic assumption. To find some reasonable values, check out [the pandapower line standard types](https://pandapower.readthedocs.io/en/v2.2.2/std_types/basic.html). Notice that the capacitance is around ~20x higher for cables compared to lines.

As a side note, long high voltage underwater cables often use DC current (HVDC) instead of AC current due to the effects of this capacitance. 

You can think of the capacitance as giving the charge more inertia - which means that AC (which is constantly changing the voltage 50 or 60 times per second) will have high losses, even when not delivering any power to loads.

marhofmann, 2020-07-01:
Hello @bergkvist,

thanks for your reply!

> My understanding is that line capacitance describes internal capacitance between the conductors of a line/cable (net.line). If you only have a single conductor, this value would be 0.

I used the term 'line-to-line capacitance' for the capacitances between two conductors  (I couldn't find the correct term in english). You are right, if there is only a single conductor this cpacitance should be 0.

> To find some reasonable values, check out the pandapower line standard types. Notice that the capacitance is around ~20x higher for cables compared to lines.

When I [checked the pandapower line types](https://pandapower.readthedocs.io/en/v2.2.2/std_types/basic.html) i also noticed, that for lines which have a single conductor the capacitance is not 0 (e.g. "NA2XS2Y **1**x95 RM/25 12/20 kV"). 
According to the [electric model of lines in pandapower](https://pandapower.readthedocs.io/en/v2.2.2/elements/line.html#electric-model) and the assumption of a symmetrical three-phase system, only a single phase of lines has to be modeled. Additionally the 'line-to-line capacitance' depends on the distance between the conductors (as you pointed out too). But for overhead lines and cables with a single conductor, the distance between the conductors is very diverse. Different towers for overhead lines or different cable trenches lead to different 'line-to-line capacitances'.

I guess i found the answer myself, but it would be nice to have it confirmed.

> You can think of the capacitance as giving the charge more inertia - which means that AC (which is constantly changing the voltage 50 or 60 times per second) will incur line power losses, even when not delivering any power to loads.

I'd just describe it like this: The line capacitance of a DC line is charged only once, shortly after the line is connected to a voltage source. On the other hand the capacitances of AC lines have to be charged in both driections 50 or 60 times per second.

bergkvist, 2020-07-04:
Oh, I see - so the line-to-line capacitance is neglected, and this is actually only the line to ground capacitance? I guess this makes sense when we look at the electric model drawing for lines (https://pandapower.readthedocs.io/en/v2.2.2/elements/line.html):

[image]

rbolgaryn, 2020-10-13:
Hi @bergkvist @maddinonline 

can this issue be closed?

marhofmann, 2021-03-05:
Hi @rbolgaryn,

if the "line capacitance" in pandapower is meant to be the line-to-earth capacitance of a line, i suggest to explicitly write it into the documentation and this issue can be closed!

Thanks

commit c2e5c995 (2021-11-16): close #828
closed 2021-11-16 by SteffenMeinecke