e2nIEE/pandapower #830: Implicitly setting bus voltage boundaries of ext_grid

friederikemeier, 2020-07-02:
Hi,

I stumbled upon a behaviour of pandapower, that I think is odd, even though I implemented it myself 😄 
The maximum and minimum bus voltages of slack buses are internally set to the slack voltage defined in `net.ext_grid.vm_pu`.

This can not be controlled by the user... I feel like this should be configurable. Because the user can set the maximum and minimum voltages in the bus table and yet this is not considered at all for the slack bus.

At the moment I am improving the PowerModels interface. I am giving the same Matpower file to the Powermodels OPF in Julia and into the pandapower-PowerModels interface and am getting different results, because of this.

I would state that the same Matpower file should lead to the same results, right?
Should this be added to the `kwargs` as a new parameter `constrain_ext_grid_vm_pu`?? And should the default value be False in order to be in line with Matpower and PowerModels?

FlorianShepherd, 2020-07-08:
Hi,
I stumbled upon this as well and had some issues (https://github.com/e2nIEE/pandapower/issues/511). 

When you implement it like this, what is the difference between a slack bus (ext_grid) and gen bus then? 

If you set the gen bus to "controllable==True" you can define limits for P, Q and Vm. The slack bus has limits for P and Q only at the moment. If you want to have a "slack" (which doesn't exist in the OPF anyway) with controllable voltage magnitudes, why not model it as a gen bus?

friederikemeier, 2020-07-10:
Exactly. We had this discussion multiple times and that is the reason why we decided to fix the voltage limits of the slack bus to the slack voltage... It is just odd if we are comparing the results of Matpower, pandapower and PowerModels... It is implicit and may be confusing to new users.. And there is no way that I can control this as a user. It's a consistency issue...

FlorianShepherd, 2020-07-10:
Ok, so if this is needed, I'd suggest keeping it consistent with the generator implementation. Currently the `controllable` parameter in the gen decides if, for example, `vm_pu` is fixed to the value in `net.gen.vm_pu` or if it must only within the boundaries of `net.bus.min_vm_pu` / `net.bus.max_vm_pu` or `net.gen.min_vm_pu` / `net.gen.max_vm_pu` (depending which of them is tighter). Instead of adding a `constrain_ext_grid_vm_pu` value, I'd add a `controllable` field to the `ext_grid`. If it is `True` the boundaries of the bus / ext_grid are considered otherwise the ext_grid values are fixed.

friederikemeier, 2020-07-10:
I agree! @lthurner what do you think?

lthurner, 2020-07-13:
Agreed

rbolgaryn, 2020-10-13:
Hi @friederikemeier @lthurner 

can the issue be closed?

friederikemeier, 2020-10-14:
All done and tested.

closed 2020-10-14 by friederikemeier