e2nIEE/pandapower #832: Bending Points

na1203132, 2020-07-02:
Hello, 

Regarding the bending points of lines in the example as shown in the image below (highlighted in red circles), are those bending points made using bus of type node or just by manipulating geodata attribute of the Line object? As I noticed that there are small circles in each bending point. This picture was brought from ieee_european_lv_asymmetric('on_peak_566') network.

Thanks in Advance. 

[image]

ascheidl, 2020-07-03:
In this grid the bending points are indeed nodes.

However, for faster load flow calculation one should instead use the possibility to model bending points via line strips in line_geodata.

na1203132, 2020-07-03:
> In this grid the bending points are indeed nodes.
> 
> However, for faster load flow calculation one should instead use the possibility to model bending points via line-strings in line_geodata.

Understood. Thank you very much for your clarification.

closed 2020-07-08 by lthurner