e2nIEE/pandapower #833: Powerflow: Jacobi's law and Efficiency

bergkvist, 2020-07-04:
## It would be nice if there was a way of getting the maximum available power for every load, along with efficiency of every load for a given network

Jacobi's law, also known as [the maximum power transfer theorem](https://en.wikipedia.org/wiki/Maximum_power_transfer_theorem) can for a given circuit be calculated by using a Thevenin equivalent simplification of a circuit.

[image]

```py
# z_s is the thevenin equivalent impedance, and z_l is load impedance.
z_s = r_s + j * x_s

# maximum power transfer happens when z_l = z_s* (complex conjugate)
z_l = r_s - j * x_s

```

What this means is that it is possible to see which loads in the network are operating close to their limits - and how much more power it is possible to add before causing the powerflow to diverge.

---

Another useful metric that comes out of a thevenin equivalent simplification is that it is easy to calculate efficiency. The efficiency is very useful in calculating what the cost of delivering power to a load is (since we can see how much is consumed in transit).

bergkvist, 2020-07-06:
### I just realized: This won't actually work as well as I had hoped
Calculating the maximum available power using the thevenin equivalent simplification makes the assumption that impedances in the rest of the network remain constant while power is added to a specific load. This is clearly not the case.

The resulting "maximum power" for a particular load would be an optimistic estimate. In order for the other loads in the network to retain their power output, they are forced to adapt their impedances (by decreasing them) as the voltage drops. This causes voltage to drop further - propagating a sort of chain reaction (which may or may not stabilize)

---

At some point, decreasing your impedance will actually also decrease your power output. This is a sign that you will not be able to extract any more power from the network (according to Jacobi's law/Thevenin equivalent simplification).

lthurner, 2020-07-08:
Have you looked at the continuation power flow method? That sounds like the direction that you are going with this (if I understood it correctly).

If so, there is a CPF implemented in MATPOWER which should also be available in PYPOWER (althourhg have not checked). It shouldn't be very difficult to connect this to pandapower.

bergkvist, 2020-08-03:
No, I haven't - but thanks for the suggestion!

continuation power flow seems to be close to what I'm looking for.

It would be great if the PYPOWER implementation was connected to pandapower.

bergkvist, 2020-10-16:
Seems like PYPOWER added continuation power flow 20 days ago! https://github.com/rwl/PYPOWER/commit/b05591d415cd515a37f8e61766458405300ac3c3

lthurner, 2020-10-19:
Cool, nice to see there is activity again in the PYPOWER repo! Are you able to look into a pandapower integration?

AnkurArohi, 2022-03-25:
Hey, maybe its a bit late for this suggestion, but you could actually do a AC Sensitivity analysis using Jacobian Matrix.

So what I am suggesting is that you could see the sensitivity of your load node to all the connected nodes to this load.
So for 1 MW increase(1 p.u., depends how have you defined it) at load node I would try to find the sensitivity / affect of this on all connected nodes.

Then I will try to see which nodes have a severity issue/ I mean where is the active power injection more than the limit of the generation if there is a generator connected or at slack bus, to see if the slack input is exceeding the measurement/normal operational limits.

But if you want to increase the pu value more than 1, you would have to do some pre-analysis/experimental tries like making the nominal loading already two times the initial and try sensitivity analysis