e2nIEE/pandapower #834: Add elements in bulk

chefPony, 2020-07-07:
Hello everyone,
adding elements one by one to an empty pandapower network which, as I understood, is currently the main approach of initialization is extremely inefficient. For example doing:

```
start = time.time()
for i in range(1000):
    pp.create_bus(pp_net, vn_kv = 0.4)
end = time.time() - start
print(end) #4.64 seconds
```
Is much slower than:
```
start = time.time()
pp_net_2 = pp.create_empty_network()
index = list(range(0, 2000, 2))
vn_kv_list = [0.4*np.random.rand(1)[0] for i in range(1000)]
pp.create_buses(pp_net_2, 1000, vn_kv_list, index= index)
end = time.time() - start
print(end) #0.14seconds
```

Currently only  the  `create_buses` function, but to really enhance performance during network initialization functions like `create_lines_from_parameters`, `create_loads`, `create_sgens `, `create_transformers_from_parameters` etc.. should be added.

PS: Also a `create_lines `function is present but its usage is too restrictive to be useful in my opinion since lines are added only if they are of the same type.

lthurner, 2020-07-08:
Yes that sounds like a good idea. Are you willing to put in a Pul Request for this? Any improvement is welcome.

chefPony, 2020-07-10:
Sure, I'll see what I can do.

closed 2020-09-01 by lthurner