e2nIEE/pandapower #836: Asymmetric Load and Unbalance Power Flow

ahhm14, 2020-07-09:
Hello!

I'm looking at the documentation for Asymmetric Load Flow (https://pandapower.readthedocs.io/en/latest/elements/asymmetric_load.html) and Unbalance Power Flow (https://pandapower.readthedocs.io/en/latest/powerflow/ac_3ph.html) 

I understand this features are included in the latest version of the library, although, is this version already avaiable? I have v2.2.2, the latest version in PyPi project. 

If not available right now, do you have an estimate of when it will be?

Thank you so much!

lthurner, 2020-07-13:
Hi, it is included in the develop version, there is a tutorial on the homepage on how to install it from github.

edxu96, 2020-07-17:
Hi, `pandapower` team:

Thanks for the excellent work. 

I have a similar problem with asymmetric lines. The task is to calculate unbalanced power flows based on data sets from [IEEE PES AMPS DSAS Test Feeder Working Group](https://site.ieee.org/pes-testfeeders/resources/). For example, in 13-bus Feeder of  1992 Test Feeder Cases, line parameters are given as:

[image]

[image]

I am wondering how I can specify phases of different kinds of lines using [`pandapower.create_line_from_parameters`](https://pandapower.readthedocs.io/en/latest/elements/line.html#pandapower.create_line_from_parameters), because they have different number of phases.

lthurner, 2020-08-03:
Single or two phase lines can not be modeled with pandapower yet, only unbalanced loads/generators in 3phase line networks.

edxu96, 2020-08-12:
May I ask why? I am doing my master graduation project. I am appreciated if you can elaborate on the difficulties with single- or V- phase laterals. Or you can recommend some literature that I can refer.

edxu96, 2021-11-25:
@gourab2009 Thanks for the reply.

But I did find a paper, `hsieh2017matrix`, later. The modeling is very similar to three-phase lines. For example:

[image]

[image]

- Hsieh, Ting‐Yen, Tsai‐Hsiang Chen, and Nien‐Che Yang (2017). “Matrix decompositions‐based approach to Z‐bus matrix building process for radial distribution systems”. In: International Journal of Electrical Power & Energy Systems 89, pp. 62–68.

heckstrahler, 2025-11-26:
Closed, multiconductor pandapower will be available soon.