e2nIEE/pandapower #837: How to check the Tap Changer Sensitivity by Pandapower?

Houshyar20, 2020-07-14:
Dears,

I'm a new learner of power system and also pandapower. I have a problem and I appreciate if you can help me.
I want to check the voltage sensitivity formula of Tap Changer for a network (e.g. cigre_network_mv(with_der="all")) by pandapower.
I selected the following inputs:
net.trafo.std_type="25 MVA 110/20 kV"
net.trafo.tap_side="lv";
net.trafo.tap_step_percent = 0.1;
net.trafo.tap_min = -20;
net.trafo.tap_max= 20;
(I left other relevant quantities as default).
By setting net.trafo.tap_pos from -20 to 20 and implementing pp.runpp(net), I did not get any change in the voltages (vm_pu & va_degree) of all buses, even by changing the tap_side to "hv".
I expected, by changing the tap_pos, that I see the voltage variations. Did I miss including other thing(s)?

Thank you very much for your comment in advance.
Best regards

wangzhenassd, 2020-07-15:
Hi @Houshyar20, you also need to define "tap_neutral" to make it work.

Houshyar20, 2020-07-15:
Hi @wangzhenassd 

Thank you for your reply. 
So, tap_neutral is in the medium position (by default) for which tap_ratio is vm_pu_1/vm_pu_2 where the indecies 1 and 2 denote, respectively, the primary and secondary sides, right?
Or, is there any arbitrary choice to select tap_neutral?

Best regards,

Houshyar20, 2020-07-16:
Hi @wangzhenassd 

Thank you for your reply. 
So, I set net.trafo.tap_neutral=0 and it worked. 🙏

Best regards,

Best regards,

closed 2020-07-16 by Houshyar20