e2nIEE/pandapower #838: DC Powerlines in OPF

pplagowski, 2020-07-15:
Hi, 
thanks for your continuous effort in maintaing pandapower!
I'm trying to run OPF simulations of a network with/without HVDC powerlines and compare the results. 

1. Is it correct, that OPF can optimize HVDC, although "DC line is only able to model one-directional loadflow for now, which is why p_mw / max_p_mw have to be > 0." (https://pandapower.readthedocs.io/en/v2.2.2/elements/dcline.html) ?
2. Can a (transmission level) power system be optimized that contains HVDC powerlines in the same "pp.create_empty_network", not being connected via create_ext_grid?
3. Does it work with both PyPower and PowerModels ACOPF and DCOPF?
4. The dcline is created with: 
- pp.create_dcline(net, from_bus=iFrom, to_bus=iTo,p_mw=1e4,
                                  loss_percent=1.2, loss_mw=25, vm_from_pu=1.01, vm_to_pu=1.01,
                                  max_p_mw=1e4, min_q_from_mvar=-10,
                                  max_q_from_mvar=10, min_q_to_mvar=-10, max_q_to_mvar=10).
- In my network (solved with PowerModels ACOPF) the res.dcline.pl_mw is similar to res.dcline.p_from_mw, which indicates that the entire power is lost in the dcline, although PowerModels returns net.OPF_converged=True. Are any of these values off (I tried also with  vm_from_pu = 1.02 and vm_to_pu = 1.01)?

gdgarcia, 2020-07-31:
Hi,

I came across the same problem raised by you in item 4.

The problem is that pandapower transforms which dcline in 2 decoupled generators to run the PowerModels optimization. This issue leads to inconsistencies. 

For example, as res_dcline.pl_mw is the sum of res_dcline.p_from_mw and res_dcline.p_to_mw for each dcline, pl_mw actually shows the value of activate power "generated" in the dcline, if it's negative.

Where the problem lays: in the function pandapower.auxiliary._add_dcline_gens:

```python
def _add_dcline_gens(net):
    from pandapower.create import create_gen
    for dctab in net.dcline.itertuples():
        pfrom = dctab.p_mw
        pto = (pfrom * (1 - dctab.loss_percent / 100) - dctab.loss_mw)
        pmax = dctab.max_p_mw
        create_gen(net, bus=dctab.to_bus, p_mw=pto, vm_pu=dctab.vm_to_pu,
                   min_p_mw=0, max_p_mw=pmax,
                   max_q_mvar=dctab.max_q_to_mvar, min_q_mvar=dctab.min_q_to_mvar,
                   in_service=dctab.in_service)
        create_gen(net, bus=dctab.from_bus, p_mw=-pfrom, vm_pu=dctab.vm_from_pu,
                   min_p_mw=-pmax, max_p_mw=0,
                   max_q_mvar=dctab.max_q_from_mvar, min_q_mvar=dctab.min_q_from_mvar,
                   in_service=dctab.in_service)
```

Quick solution: tying the active power limits of the generators to their nominal values. 
This makes the dclines not "dispatchable" by the PowerModels, but it's fast and consistent.

```python
def _add_dcline_gens(net):
    from pandapower.create import create_gen
    for dctab in net.dcline.itertuples():
        pfrom = dctab.p_mw
        pto = (pfrom * (1 - dctab.loss_percent / 100) - dctab.loss_mw)
        pmax = dctab.max_p_mw
        create_gen(net, bus=dctab.to_bus, p_mw=pto, vm_pu=dctab.vm_to_pu,
                   min_p_mw=pto, max_p_mw=pto, max_q_mvar=dctab.max_q_to_mvar,
                   min_q_mvar=dctab.min_q_to_mvar, in_service=dctab.in_service)
        create_gen(net, bus=dctab.from_bus, p_mw=-pfrom, vm_pu=dctab.vm_from_pu,
                   min_p_mw=-pfrom, max_p_mw=-pfrom,
                   max_q_mvar=dctab.max_q_from_mvar, 
                   min_q_mvar=dctab.min_q_from_mvar, 
                   in_service=dctab.in_service)
```

A better fix would be: modeling the HVDCs as dclines inside PowerModels.

friederikemeier, 2020-12-02:
Hi @gdgarcia ! Indeed, we only have a simplified model for the DC lines that doesn't convert to "real controllable" DC lines in powermodels. We would appreciate an update of our powermodels interface to be able to cope with this 😄

patripla, 2021-01-21:
Hi folks,
I'm trying to get dclines running in an OPF with pypower as solver (powermodels has still the interface and cost curve limitations). 
A network is created in the file attached, where line 5 can be an AC or DC line. It does converge with an AC line, but it doesn't with a DC line. Pips however starts iterating as can be seen in PyPower (ppci) System Summary attached. 
What am I doing wrong in this small DC line example?
I would very much appreciate if anyone can help with a hint or provides a working example of OPF with pypower and a DC line.

[DC_line.py.txt](https://github.com/e2nIEE/pandapower/files/5849730/DC_line.py.txt)
[Log_ACline.txt](https://github.com/e2nIEE/pandapower/files/5849732/Log_ACline.txt)
[Log_DCline.txt](https://github.com/e2nIEE/pandapower/files/5849733/Log_DCline.txt)

vogt31337, 2025-03-19:
Closed due to inactivity.

closed 2025-03-19 by vogt31337