e2nIEE/pandapower #839: KeyError: 'bus'

Alwinator, 2020-07-16:
https://github.com/jhmenke/grid2op_pp_baseline/issues/5#issuecomment-659296433

FlorianShepherd, 2020-07-17:
There is a problem with the powermodels update and the pandapower interface. Could you install the pandapower develop version from Github and try? This may work I guess

Alwinator, 2020-07-20:
Just tried it with the latest version from Github. Unfortunately the problem was **not** solved.

FlorianShepherd, 2020-07-20:
Ok, what is the convergence status of powermodels?

If the model is infeasible, there is no solution and "bus" cannot be read from the results. I just realized there is no exception for this within a time series calculation. If you enable verbosity in the powermodels output you should get something like
`Model may be infeasible or unbounded.`

Alwinator, 2020-07-20:
Where can I enable verbosity in the powermodels?

FlorianShepherd, 2020-07-20:
when calling `pp.runpm` you can set 'pm_log_level` to `1` for example:

https://pandapower.readthedocs.io/en/v2.2.2/opf/powermodels.html?highlight=runpm(#pandapower.runpm 

Then you should get more information about the solver

Alwinator, 2020-07-20:
```
┌ Warning: `with_optimizer` is deprecated. Adapt the following example to update your code:
│ `with_optimizer(Ipopt.Optimizer, max_cpu_time=60.0)` becomes `optimizer_with_attributes(Ipopt.Optimizer, "max_cpu_time" => 60.0)`.
│   caller = ip:0x0
└ @ Core :-1
Losses avoided: 1.329 MW (max. loading: 0.0% [-0.0%] / Trafo: 60.7% [-4.5%])
OPF wants to disconnect: pp_[29, 31, 34, 43, 44] / connect: pp_[] 
Observation: Line 44 out of service!
WARNING: replacing module PP2PM.
┌ Warning: NLP couldn't be solved to optimality
└ @ Juniper ~/.julia/packages/Juniper/Hm6I1/src/fpump.jl:393
Losses avoided: 1.063 MW (max. loading: 0.0% [-0.0%] / Trafo: 61.8% [-4.3%])
OPF wants to disconnect: pp_[] / connect: pp_[] 
Observation: Line 44 out of service!
WARNING: replacing module PP2PM.
┌ Warning: NLP couldn't be solved to optimality
└ @ Juniper ~/.julia/packages/Juniper/Hm6I1/src/fpump.jl:393
Losses avoided: 1.033 MW (max. loading: 0.0% [-0.0%] / Trafo: 57.9% [-9.3%])
OPF wants to disconnect: pp_[5, 6, 13, 29, 30, 31, 34, 43] / connect: pp_[] 
Observation: Line 43 out of service!
Observation: Line 44 out of service!
WARNING: replacing module PP2PM.
┌ Warning: NLP couldn't be solved to optimality
└ @ Juniper ~/.julia/packages/Juniper/Hm6I1/src/fpump.jl:393
Losses avoided: 0.931 MW (max. loading: 0.0% [-0.0%] / Trafo: 57.7% [-10.0%])
OPF wants to disconnect: pp_[5, 6, 29, 30, 31, 34] / connect: pp_[] 
Observation: Line 34 out of service!
Observation: Line 43 out of service!
WARNING: replacing module PP2PM.
┌ Warning: NLP couldn't be solved to optimality
└ @ Juniper ~/.julia/packages/Juniper/Hm6I1/src/fpump.jl:393
┌ Warning: Only almost solved
└ @ Juniper ~/.julia/packages/Juniper/Hm6I1/src/BnBTree.jl:89
Losses avoided: 0.742 MW (max. loading: 0.0% [-0.0%] / Trafo: 63.8% [-4.8%])
OPF wants to disconnect: pp_[] / connect: pp_[44] 
Observation: Line 34 out of service!
Observation: Line 43 out of service!
WARNING: replacing module PP2PM.
┌ Warning: NLP couldn't be solved to optimality
└ @ Juniper ~/.julia/packages/Juniper/Hm6I1/src/fpump.jl:393
Losses avoided: 0.666 MW (max. loading: 0.0% [-0.0%] / Trafo: 64.5% [-4.9%])
OPF wants to disconnect: pp_[] / connect: pp_[] 
Observation: Line 34 out of service!
WARNING: replacing module PP2PM.
--------

ERROR: Your agent could not be run. 
It will probably fail on codalab.
Here is the information we have:

INFO: the backend used in codalab is "LightSimBackend" which is not available on your system. Please visit "https://github.com/BDonnot/lightsim2grid" to install it. NB results difference between native backend (PandaPowerBackend) and LightSimBackend are below 1e-5 in most cases.
input dir: /home/alwin/PycharmProjects/l2rpn-challenge/submission/input_data_local/
output dir: utils/last_submission_results/res
program dir: /home/alwin/PycharmProjects/l2rpn-challenge/submission/utils/ingestion_program_local/
submission dir: /tmp/tmpw37mtgjv
input content ['prods_charac.csv', '__pycache__', 'grid_layout.json', 'difficulty_levels.json', '.TODO', 'chronics', 'config.py', 'grid.json']
output content []
program content ['metadata', 'config_local.json', 'ingestion.py']
Content received by codalab: ['metadata', 'submission']

INFO: No custom reward for the assessment of your agent will be used.

INFO: No custom other_rewards for the assessment of your agent will be used.

WARNING: Could not load lightsim2grid.LightSimBackend, falling back on PandaPowerBackend

[info | PowerModels]: removing 1 cost terms from generator 4: Any[1.0, 0.0]
[info | PowerModels]: removing 1 cost terms from generator 1: Any[1.0, 0.0]
[info | PowerModels]: removing 3 cost terms from generator 12: Any[]
[info | PowerModels]: removing 1 cost terms from generator 20: Any[1.0, 0.0]
[... 448 lines cut ...]
    env_seeds=env_seeds, max_iter=max_iter, agent_seeds=agent_seeds)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/grid2op/Runner/Runner.py", line 819, in _run_sequential
    max_iter=max_iter)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/grid2op/Runner/Runner.py", line 592, in run_one_episode
    pbar=pbar, env_seed=env_seed, max_iter=max_iter, agent_seed=agent_seed)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/grid2op/Runner/Runner.py", line 694, in _run_one_episode
    act = agent.act(obs, reward, done)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/l2rpn_baselines/PandapowerOPFAgent/PandapowerOPFAgent.py", line 175, in act
    self.opf_type.lower(), logger=self.logger)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/l2rpn_baselines/PandapowerOPFAgent/pp_functions.py", line 55, in run_opf
    pp.runpm_ots(grid, pm_nl_solver="ipopt", pm_model="ACPPowerModel", pm_log_level=1)  # PowerModels.jl OPF
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/pandapower/runpm.py", line 320, in runpm_ots
    _runpm(net)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/pandapower/opf/run_powermodels.py", line 38, in _runpm
    read_pm_results_to_net(net, ppc, ppci, result_pm)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/pandapower/converter/powermodels/from_pm.py", line 18, in read_pm_results_to_net
    result, multinetwork = pm_results_to_ppc_results(net, ppc, ppci, result_pm)
  File "/home/alwin/miniconda3/envs/l2rpn_challenge/lib/python3.7/site-packages/pandapower/converter/powermodels/from_pm.py", line 51, in pm_results_to_ppc_results
    for i, bus in sol["bus"].items():
KeyError: 'bus'
------------------------------------
      End Detailed error Logs       
------------------------------------

```

FlorianShepherd, 2020-07-20:
Thank you for the report. The solver does, indeed, not converge. I can add an exception as a fix and check if "bus" is in sol, otherwise "OPF not converged" will be raised.

FlorianShepherd, 2020-07-20:
This should do the trick: https://github.com/FlorianShepherd/pandapower/commit/21452efcbfd294d93342eff11094aff5bcf140e6 ?

Just pull from my develop branch:
https://github.com/FlorianShepherd/pandapower/

Could you try this? Be aware that it will raise an exception now "pp.OPFNotConverged". You can try except this with:
```
from pandapower.optimal_powerflow import OPFNotConverged
try:
    pp.runpm(net)
except OPFNotConverged:
   pass
```

Alwinator, 2020-07-21:
Still, the same exception. I installed pandapower via `pip install git+https://github.com/e2nIEE/pandapower.git@develop`

Alwinator, 2020-07-21:
Thank you. I think the problem in pandapower is now fixed. But, still there is not everything working. I will continue the Issue from [here](jhmenke/grid2op_pp_baseline#5 (comment)).

commit f900462b (2020-07-31): Merge pull request #844 from FlorianShepherd/develop

fixes  #839 and added Powermodels OPFNotConverged expection if Powermodels OPF fails
closed 2020-07-31 by FlorianShepherd