e2nIEE/pandapower #847: sensitivity analysis

Homam94, 2020-08-02:
Hello,
How I can do sensitivity analysis of a distrubition network for example with the help of Ptdf in Pandapower?
Is there any functions that I can use in this case?

wangzhenassd, 2020-08-03:
Hi @Homam94, 
The ptdf/lodf requires the ppci data structure, which can be aquired by pd2ppc function
you can check out this #784 for the usage of the internal data structure.

closed 2020-08-07 by wangzhenassd

marinaffc, 2021-11-24:
Hi @wangzhenassd , 

thank you for your comment. I have seen that #784 issue, however, same as @Homam94 I am not sure how to use the internal data structure to calculate PTDF. Do you have any available reference?

Thank you very much!

rbolgaryn, 2021-11-27:
There is a PTDF function in pandapower/pypower. You can give it a try and see if it is helpful for your use-case
Roman