e2nIEE/pandapower #851: [Feature/improvement] Faster/more scalable graph algorithms with GraphBLAS

bergkvist, 2020-08-03:
# Recent radical innovation

There has been some radical innovation in graph processing the last few years

GraphBLAS efficiently represents and operates on graphs as sparse matrices. It provides building blocks for creating performant - and highly parallel/scalable graph algorithms. 

I think pandapower could benefit greatly in some ways from utilizing GraphBLAS (in terms of performance/scalability)

# Potential use cases

### Connected Components
FastSV from the litterature review is ~20x faster than the NetworkX implementation for a network with ~300,000 buses/lines - and that is in single threaded, blocking mode. The algorithm is close to being [embarrassingly parallel](https://en.wikipedia.org/wiki/Embarrassingly_parallel)

### Short Circuit Analysis
Since graph edges in GraphBLAS can be complex numbers - this might be a perfect fit for circuit analysis/simplifications.

### Other Graph related problems
I'm guessing there are probably more graph-related things in pandapower that could become faster and more scalable with GraphBLAS.

## Litterature review

[image]

#### [(2017) SuiteSparse:GraphBLAS: graph algorithms in the language of linear algebra](https://github.com/DrTimothyAldenDavis/GraphBLAS/blob/stable/Doc/GraphBLAS_UserGuide.pdf)

GraphBLAS takes advantage of sparse matrices, as well as the fact that graph operations can be effectively represented by linear algebra. The implementation is written in C.

- [(_Source code/GitHub_)](https://github.com/DrTimothyAldenDavis/GraphBLAS)

---

#### [(Jan 2019) LAGraph: library plus a test harness for collecting algorithms that use the GraphBLAS](http://faculty.cse.tamu.edu/davis/GraphBLAS_files/lagraph-grapl19.pdf)

While GraphBLAS provides the linear algebra primitives for building more complex graph algorithms - LAGraph collects efficient algorithms built on top of GraphBLAS.

- [(_Source code/GitHub_)](https://github.com/GraphBLAS/LAGraph)

---

#### [(Aug 2019) GraphBLAST: A High-Performance Linear Algebra-based Graph Framework on the GPU](https://arxiv.org/abs/1908.01407)

GraphBLAST is a GPU implementation of GraphBLAS

- [(_Source code/GitHub_)](https://github.com/gunrock/graphblast)

---

#### [(Oct 2019) FastSV: a distributed memory connected-component algorithm with fast convergence](https://deepai.org/publication/fastsv-a-distributed-memory-connected-component-algorithm-with-fast-convergence)

Significantly faster alternative to [`pandapower.topology.connected_components`](https://pandapower.readthedocs.io/en/latest/topology/searches.html#pandapower.topology.connected_components)

- [(_Source code/GitHub_)](https://github.com/GraphBLAS/LAGraph/blob/master/Source/Algorithm/LAGraph_cc_fastsv.c)

---

#### [(Jun 2019) pygraphblas: GraphBLAS for Python](https://github.com/michelp/pygraphblas/blob/master/pygraphblas/demo/Introduction-to-GraphBLAS-with-Python.ipynb)

Python binding for GraphBLAS

- [(_Source code/GitHub_)](https://github.com/michelp/pygraphblas)

---

FlorianShepherd, 2020-08-06:
Hi @bergkvist ,

nice idea. I also  wrote an interface for the graph tool a while ago: 
https://github.com/e2nIEE/pandapower/blob/4a3dc519ab33a5de9baf6fe75d3156bdb432861c/pandapower/topology/graph_tool_interface.py

since networkx was too slow for my case as well. There is just some licensing issue and also it doesn't work so easily on windows. I'm not sure about apache 2.0 and GraphBLAS but it looks  definitely  cool

Best
Florian

lthurner, 2020-08-10:
Looks good. One question: how easy/difficult can GraphBLAS be installed on Unix/Windows respectively?

bergkvist, 2020-08-10:
On Ubuntu/Debian: https://packages.debian.org/sid/libgraphblas3
```sh
sudo apt install libgraphblas3
```

## Building from source on Linux (requires CLang)
On a 2-core laptop-cpu this can take as much as a couple of hours, so this is probably something you'd want to avoid doing on installation. Precompiled binaries/shared libraries are around 50MB in size on Linux.
```sh
# The "stable" branch is default, so no need to switch branch after cloning
git clone https://github.com/DrTimothyAldenDavis/GraphBLAS
cd GraphBLAS
make JOBS=8
sudo make install
```

After running `sudo make install`, the following files will appear in the file system:
```
/usr/local/lib/
  - libgraphblas.so.3.3.3
  - libgraphblas.so.3
  - libgraphblas.so

/usr/local/include/
  - GraphBLAS.h
```

I've created a Docker image that contains shared libraries/header files (and nothing else) for GraphBLAS and LAGraph here: https://hub.docker.com/r/bergkvist/lib-lagraph

---

Regarding Windows: I have no idea - so this is something to look into.

bergkvist, 2020-08-13:
Just discovered that GraphBLAS became part of Anaconda 21 days ago. This should probably make it easier to install on Windows.
https://anaconda.org/conda-forge/graphblas

[image]

bergkvist, 2020-08-25:
Update: [pygraphblas](https://github.com/michelp/pygraphblas) is now available on [conda-forge](https://github.com/conda-forge/pygraphblas-feedstock), since 19 hours ago.

lthurner, 2020-08-27:
Cool, then installation shouldn't be an issue.

About the potential use cases:

> **Connected Components**
> FastSV from the litterature review is ~20x faster than the NetworkX implementation for a network with ~300,000 buses/lines - and that is in single threaded, blocking mode. The algorithm is close to being embarrassingly parallel

Its true that we have a connected component based on networkx in the topology. However, inside the power flow, we use scipy adjancy matrix search to check for disconnected components: https://github.com/e2nIEE/pandapower/blob/1e7842bf9c8b9555857afa22d21a5d0feabe8bfa/pandapower/auxiliary.py#L404-L432 Since scipy implements this search in C++ libraries internally, its also pretty fast compared to networkx. 

> *Short Circuit Analysis*
>Since graph edges in GraphBLAS can be complex numbers - this might be a perfect fit for circuit analysis/simplifications. The complex number of an edge can represent its impedance.

With a matrix based solution you can compute short-circuit currents for faults at all buses in parallel. I doubt that doing a graph search for each bus will outperform a matrix based approach to be honest. Also it will only work in a radial network and not in a meshed one. So I doubt this a good fit for graph based analysis in general.

rbolgaryn, 2020-10-13:
Hi @bergkvist @lthurner @FlorianShepherd 

any updates?

bergkvist, 2020-10-16:
Seems related (considering it depends on GraphBLAS/SuiteSparse): https://github.com/BDonnot/lightsim2grid