e2nIEE/pandapower #854: #OPF: runopp() for all time steps

TiagoTorres1997, 2020-08-04:
Hi, 
I was wondering how i can run the function ```pp.runopp(net) ```for all the  timesteps that i want (e.g. 1 year) and not only one operation point of the network?

I can only apply the ```pp.runopp(net) ``` for a specific time step and see the results from the network at one operation point/time step... So, to see the results from multiples hours/timesteps of an year i have to run the multiple OPF's one by one. 

So until now, what I´m doing is "passing" the values (p_mw and q_mvar) of sgens and loads from a specific timestep  to the network and running the OPF, like this:

```ruby
net.load.p_mw = ow.np_results["load.p_mw"][12,:]
net.load.q_mvar = ow.np_results["load.q_mvar"][12,:]

net.sgen.p_mw = ow.np_results["sgen.p_mw"][12,:]
net.sgen.q_mvar = ow.np_results["sgen.q_mvar"][12,:]

#OPF
net.load['max_p_mw'] = net.load['p_mw']* net.load['scaling']
net.load['min_p_mw'] = net.load['p_mw']*0

net.load['max_q_mvar']= net.load['q_mvar']
net.load['min_q_mvar']= net.load['q_mvar']*0

net.sgen['max_p_mw']= net.sgen['p_mw']
net.sgen['min_p_mw']= net.sgen['p_mw']*0

net.sgen['max_q_mvar']= net.sgen['q_mvar']
net.sgen['min_q_mvar']= net.sgen['q_mvar']*0

for i in net.load.index:
    pp.create_poly_cost(net, i, 'load', cp1_eur_per_mw=-1300)
    
for t in net.sgen.index:
   pp.create_poly_cost(net, t, 'sgen', cp1_eur_per_mw=-95)
      
net.load['controllable'] = True
net.sgen['controllable'] = True

net.bus['min_vm_pu'] = 0.9
net.bus['max_vm_pu'] = 1.1

net.ext_grid['min_vm_pu']=1
net.ext_grid['max_vm_pu']=1

net.trafo['max_loading_percent'] = 120.
net.line['max_loading_percent'] = 110.

pp.runopp(net,verbose=True)
```

The funny thing is that the function `` run_timeseries (net) `` runs multiple ac power flows, and something similar but for multiple optimal power flows  i think is the  ideal solution to my problem.

Until this moment, i don´t know any other way to carry on multiple OPF´s that the one i show in the previous code... 

Do you have any suggestions about how to do it?

Best regards,
Tiago

FlorianShepherd, 2020-08-06:
Hi Tiago,

you can do that and it's already implemented, but a little bit 'hidden'. See here:
https://pandapower.readthedocs.io/en/v2.2.2/timeseries/run_function.html

> kwargs - Keyword arguments for run_control and runpp. If “run” is in kwargs the default call to runpp() is replaced by the function kwargs[“run”]

So if you want to call runopp() instead of runpp() in a time series loop, you have to call:

```
run_timeseries(net, run=pp.runopp)
```
Best,
Florian

TiagoTorres1997, 2020-08-07:
Hi @FlorianShepherd,

Thank you  for clarifying the matter in question. I implemented the function and worked!

Best regards,
Tiago

closed 2020-08-07 by TiagoTorres1997