e2nIEE/pandapower #856: Print a spesific bus

Homam94, 2020-08-05:
Hello everyone,
I have a quastion about the printing of a single bus, for example by the function print(net.bus,head()) I can print the first 5 Elements of the buses but how I can print a spesific bus for examplte the bus with the index = 0 ?

jkisse, 2020-08-06:
Hello Homam94,
you can print a specific bus with the pandas .loc-command. For example, to print the bus with the ID 0:
`net.bus.loc[0]`. This works for a list of indices or an array with boolean elements as well.
If you want to retrieve a specific entry of a dataframe, based on its position (regardless of the ID), you can use .iloc - it is faster than .loc. For example, if you want to print the 3rd bus in the bus-dataframe, you can use `net.bus.iloc[2]`.

More information can be found in the pandas-documentation [here](https://pandas.pydata.org/pandas-docs/stable/reference/api/pandas.DataFrame.loc.html) .

FlorianShepherd, 2020-08-06:
and also you can use `df.iat[]` and `df.at[]`  instead if iloc and loc, which is even faster (at least in pandas < 1.0)

closed 2020-08-07 by FlorianShepherd