e2nIEE/pandapower #858: Issue with the state estimation of the 33-bus system

RebbahD, 2020-08-07:
Dear All, 

I am currently trying to use pandapower, to accomplish the state estimation of the 33 bus system using the network from pandapower.networks: 

import pandapower.networks as pn
import pandapower as pp
net = pn.case33bw()
pp.runpp(net)

I have created 65 measurements using : 
pp.create_measurement(net, "p", "line", -0.090044, -0.090044*0.03, element=20, side="from")
as an example. 

then I have used : 
from pandapower.estimation import estimate
success = estimate(net, init="flat")
print(success)

which always returns: False with this error: 

C:\Users\johnw\anaconda3\lib\site-packages\scipy\sparse\linalg\dsolve\linsolve.py:206: MatrixRankWarning: Matrix is exactly singular
  warn("Matrix is exactly singular", MatrixRankWarning)
Estimation failed! Pandapower network failed to update!

Please help me resolve this issue, as I have tried every solution possible.

Best Regards,
Dhiaa.

wangzhenassd, 2020-08-07:
I think you should consider add some more measurements, to reach the minimum number of measurements (2n-k) does not guarantie the wls algorithm to solve it. Measurements may have different impact on the convergence of the algorithm, e.g. if you have at the same time p_from,q_from,p_to,q_to of one single line, this configuration does not have the effect of 4 independent measurements.

RebbahD, 2020-08-08:
Hello, I do appreciate the reply, however, I am confused about the measurements side. I basically used : 

print(net.res_bus)
print(net.res_line)
print(net.line)

to get the data and used it as measurements 

is that a correct practice? is there a way to get the measurements in a better way for the 33 bus system 

I am using : 

import pandapower.networks as pn

net = pn.case33bw()
pp.runpp(net)

what should I do to get the measurements data to use it for WLS state estimation?

wangzhenassd, 2020-08-08:
Hi, the following code should do the trick, but it seems to me there is a strange bug regarding the out-of-service lines. I will fix it soon.
```python
import pandapower as pp
import pandapower.networks as nw
from pandapower.estimation import estimate

net = nw.case33bw()
pp.runpp(net)

for line_ix in net.line.index.to_numpy():
    pp.create_measurement(net, "p", "line", net.res_line.at[line_ix, "p_from_mw"],
                          0.01, element=line_ix, side="from" )
    pp.create_measurement(net, "q", "line", net.res_line.at[line_ix, "q_from_mvar"],
                          0.01, element=line_ix, side="from" )
    
for bus_ix in net.bus.index.to_numpy():
    pp.create_measurement(net, "v", "bus", net.res_bus.at[bus_ix, "vm_pu"],
                          0.01, element=bus_ix )
    
estimate(net)
```

RebbahD, 2020-08-08:
Hello dear Wang,

I am still facing the error of : 

IndexError: index 32 is out of bounds for axis 0 with size 32

is there a solution for that?

RebbahD, 2020-08-09:
The issue is at: 

  File "C:\Users\johnw\Desktop\estimationtest.py", line 20, in <module>
    estimate(net)

  File "C:\Users\johnw\anaconda3\lib\site-packages\pandapower\estimation\state_estimation.py", line 82, in estimate
    return se.estimate(v_start=v_start, delta_start=delta_start,

  File "C:\Users\johnw\anaconda3\lib\site-packages\pandapower\estimation\state_estimation.py", line 257, in estimate
    self.net, self.ppc, self.eppci = pp2eppci(self.net, v_start=v_start, delta_start=delta_start,

  File "C:\Users\johnw\anaconda3\lib\site-packages\pandapower\estimation\ppc_conversion.py", line 588, in pp2eppci
    ppci = _add_measurements_to_ppci(net, ppci, zero_injection)

  File "C:\Users\johnw\anaconda3\lib\site-packages\pandapower\estimation\ppc_conversion.py", line 246, in _add_measurements_to_ppci
    branch_append[ix_from, P_FROM] = meas_from.value.values

IndexError: index 32 is out of bounds for axis 0 with size 32

what should I do here please?

wangzhenassd, 2020-08-09:
That is the bug I am trying to repair. The code should work after #859 merged.

RebbahD, 2020-08-09:
The issue is, I am working on my thesis project which is only on the case33, should I wait for the solution?

wangzhenassd, 2020-08-09:
Then yes, it should be merged in 2 or 3 days

RebbahD, 2020-08-09:
Perfect, I will be waiting.

lthurner, 2020-08-11:
Merged on develop, release coming soon.

closed 2020-08-11 by lthurner