e2nIEE/pandapower #862: Expose all calculated short circuit branch currents

chefPony, 2020-08-11:
Currently the functions `calc_sc` only provide the max or min values of the current flowing in a particular branch, meanwhile `calc_single_sc `provides currents for all branches but with a single fault location specified. But in many applications such as fault protection optimal placement, tuning, lgrid reliability analysis the short circuit currents flowing in every branch for every possible fault bus location can be useful.

The expected output would be matrix-like of the sort:
```
                 bus_0    ....     bus_j     ....       bus_m

branch_0       i_ka_00   ....     i_ka_0j       ...    i_ka_0m       
....
branch_i        i_ka_i0   ....     i_ka_ij       ...    i_ka_im            
...
branch_n       i_ka_n0   ....     i_ka_nj       ...    i_ka_nm       
```

Where `i_ka_ij` is the current of branch element `i` when the fault occurs at bus `j`. It seems like this result is actually computed internally in the existing pandapower routines, but is not exposed to the user. It would be nice if it could be retrieved (if some option is specified) or to have a separate function that return this information.

lthurner, 2020-08-11:
I agree and I had also already thought about this, but no time yet to implement it. But as you said, all routines are already in place to do it, maybe its not that much effort.

It is a little tricky to put the results into a DataFrame. Usually we have the parameter name (string) as the column, and the element index (integer) as the index. However, columns can't be integers, so we would probably have to do it the way you outlined it with the "bus_<bus_index>" string. Its not in line with the rest of the pandapower conventions, but I don't really see a better way either.

One easy first step in this direction would be to simply store the matrix during the calculation in the "internal" dictionary? Then it could be accessed afterwards. However, the user would have to use the lookups from the pandapower indices to the internal data structure, so its probably not that convenient.

chefPony, 2020-08-12:
Storing the matrix in the "internal" dictionary could be a good start.
As a next step to put the results in a DataFrame, two ideas came in my mind are:

- Maintaining today `res_line_sc` table format but having a list of values as elements, something like:
```
#res_line_sc:
                       ikss_ka                        ith_ka                    ip_ka

0      [ikss_00, ikss_01, ..., ikss_0m]    [ith_00, ith_01, ..., ith_0m]   [ip_00, ip_01, ..., ip_0m]
1
...
n
```
where each list consist of `m` elements and j-th entry would be the current for a fault at bus j. So that if the user want to access ikss_ka for line_i for a fault at bus_j the syntax would look something like: `res_line_sc['ikss_ka'][i][j]`

- An alternative could be to use pandas multindex feature, but this would have a different format from what is currently implemented:
```
#res_line_sc:
line   bus         ikss_ka         ith_ka      ip_ka

0        0          ikss_00         ith_00      ip_00
-        1          ikss_01         ith_01      ip_01 
-        ...
-        m          ikss_0m         ith_0m      ip_0m
1
...
n
```

lthurner, 2020-08-13:
> An alternative could be to use pandas multindex feature, but this would have a different format from what is currently implemented:

I like this idea very much, its the cleanest solution and most compatible to all other result tables

lthurner, 2020-08-27:
Great work, this looks very nice all around! I have merged your changes and added a couple of tests that compare the single_sc function against the new multi-branch functions #899. The results check out 👍 

In the PR you also mentioned res_trafo3w, but there is no test that includes a traof3w and with a quick check it seems res_trafo3w_sc is empty. 3W transformers are not yet implemented yet, correct?

chefPony, 2020-08-27:
They are implemented but not tested. I think overall the functionality is there but the testing it's a bit lacking, maybe they are worth extending in the future.

SteffenMeinecke, 2021-11-11:
@chefPony I guess the the three latest commits are not included to pp since the merge has been before. Are they relevant? If yes, could you please raise a second PR? What about the testing functionality? Could you provide it or can we close this issue without it?

chefPony, 2021-11-12:
Hi @SteffenMeinecke , to me it seems that the last three commit are included in pp as well [](https://github.com/e2nIEE/pandapower/pull/890). Unfortunately, I do not have a reference result for the trafo3w to use for testing and I don't have much time lately, so I guess you can close the issue.

closed 2021-11-12 by SteffenMeinecke