e2nIEE/pandapower #866: Per unit calculation of transformer impedance

skababji, 2020-08-13:
The per unit calculation of transformer impedance indicated in https://pandapower.readthedocs.io/en/v2.3.0/elements/trafo.html does not seem to yield the correct result. It seems there is something  wrong with the indicated formulas. However, upon checking transformers' corresponding branches in ._ppc, impedances seem okay.

Per unit value of trafo impedance may be simply calculated by:
z_pu=['vk_percent'] of transformer/100)*(['sn_mva'] of system/['sn_mva'] of transformer)
The above will yield results more in line with matpower case studies (scaled by 100 due to differences in system base mva).

lthurner, 2020-08-14:
>  z_pu=['vk_percent'] of transformer/100)*(['sn_mva'] of system/['sn_mva'] of transformer)

You right. The formula in the documentation was created when sn_mva of the system was still fixed internally to 1 MVA and transformer rated power was given in kVA. Seems like the formula was not correctly updated in the documentation.

The only thing that has to be changed though is replacing "1000" by net.sn_mva, right?

skababji, 2020-08-14:
Yes. This (i.e. replacing "1000" by "net.sn_mva") needs to be done for both z_k and r_k. Then the last equations about using Z_N and Z_ref,trafo  need to be deleted. 
I guess the same logic will apply to the admittance but I did not check it.

commit 2add154b (2020-08-17): Merge pull request #872 from lthurner/develop

fix transformer equations Closes #866
commit 504d37e5 (2020-08-17): fix transformer equations Closes #866
closed 2020-08-17 by lthurner