e2nIEE/pandapower #867: add LV grid for town in MV Oberrhein region

jkisse, 2020-08-14:
The low voltage grid from this paper https://www.mdpi.com/1996-1073/13/16/4052 should be added to the networks-module.

rbolgaryn, 2020-10-13:
Hi @jkisse ,

what is the current status of the issue?

jkisse, 2020-10-14:
I can do this next week.

jkisse, 2020-12-09:
I had to do some modifications regarding the substations (add more) so the trafos do not align 100 % with the MV Oberrhein network. 
If anyone is interested in the network file, please let me know.

SteffenMeinecke, 2021-11-11:
> If anyone is interested in the network file, please let me know.

@jkisse So the LV grid from the paper will be not included to pandapower and we can close this issue? Would be sad for the paper and the community if we don't provide it.

pawellytaev, 2024-11-25:
This is about the Schutterwald grid right? Thus, I'm closing this issue since it's already in pandapower

closed 2024-11-25 by pawellytaev