e2nIEE/pandapower #869: Bug in rundcpp failed to initialize result table

wangzhenassd, 2020-08-14:
```python
import pandapower as pp
import pandapower.networks as nw

net = nw.case57()
pp.rundcpp(net)    
net.pop("res_gen")

# Problem by executing this
pp.rundcpp(net)

# No problem doing this
pp.runpp(net) 
```

Can anyone recreate this? I tries pp 2.2.2, pp 2.3.0.

wangzhenassd, 2020-08-17:
@FlorianShepherd can you try this case? Is this reproducible?

FlorianShepherd, 2020-08-17:
@wangzhenassd I can confirm. I get this error:
```
  File "/home/florian/python_script/temp/dcbug.py", line 9, in <module>
    pp.rundcpp(net)
  File "/home/florian/pandapower/pandapower/run.py", line 267, in rundcpp
    _powerflow(net, **kwargs)
  File "/home/florian/pandapower/pandapower/powerflow.py", line 52, in _powerflow
    verify_results(net)
  File "/home/florian/pandapower/pandapower/results.py", line 73, in verify_results
    if len(net[element]) != len(net[res_element]):
KeyError: 'res_gen'
```
I'll try to find a  see how to fix this

closed 2020-08-17 by FlorianShepherd

wangzhenassd, 2020-08-17:
Thank you! @FlorianShepherd  It works now.

commit d64dc102 (2020-08-17): Merge pull request #870 from FlorianShepherd/develop

fixes #869 missing res_gen if rundcpp is called
commit e4c71bd6 (2020-08-17): fixes #869