e2nIEE/pandapower #871: Fail in test.run_all_test

hitchenzhe, 2020-08-17:
Hello,

I'm  the new to use pandapower. And I run following command, it failed. How to fix it? Thanks

`import` pandapower.test
`pandapower.test.run_all_tests()`

The terminal say:
.local/lib/python3.6/site-packages/pandapower/test/control/test_control.py::test_multiple_levels
  /home/zhe/.local/lib/python3.6/site-packages/numpy/core/_asarray.py:83: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray
    return array(a, dtype, copy=False, order=order)

-- Docs: https://docs.pytest.org/en/stable/warnings.html
==================== short test summary info ======================
FAILED ../../.local/lib/python3.6/site-packages/pandapower/test/control/test_control.py::test_multiple_levels - ValueError: setting an array element with a sequence.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
==================== 1 failed, 123 passed, 1 skipped, 2 xfailed, 88 warnings in 199.39s (0:03:19) =============

lthurner, 2020-08-17:
This should be fixed in the newest version pandapower 2.3.0 that has been released last week.

What version of pandapower are using and how did you install it?

hitchenzhe, 2020-08-17:
I use 'pip3 install pandapower' this morning, I think it's the latest version.

lthurner, 2020-08-18:
Strange - can you please tell me which versions of pandapower, pandas and numpy you are using. Either check in your pip list or directly in python with:

> import pandapower
> print(pandapower.__version__)

hitchenzhe, 2020-08-18:
> Strange - can you please tell me which versions of pandapower, pandas and numpy you are using. Either check in your pip list or directly in python with:
> 
> > import pandapower
> > print(pandapower.**version**)

I think this may be the problem, when I run version command:

numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)

<module 'packaging.version' from '/home/zhe/.local/lib/python3.6/site-packages/packaging/version.py'>

lthurner, 2020-08-19:
The error that you copied above is an error that popped up with pandas 1.1, I don't think it has anything to do with numba. Can you please still name your versions for numpy, pandas, pandapower?

Hexelena, 2020-08-19:
I got the same error with 

```python==3.7.7```
```pandas==1.1.0```
```numpy==1.19.1```

I setup a completely new venv to rule out any other errors.

Downgrading pandas to 1.0.5 solved the problem, just leaving the error from #876.

lthurner, 2020-08-19:
Thats weird, the tests on Travis CI pass with exactly the same combination of pandas & numpy: https://travis-ci.org/github/e2nIEE/pandapower/jobs/719257497

Hexelena, 2020-08-19:
Though i don't get the error with pandapower 2.3.1 (although pandas and numpy are still at 1.1.0 and 1.19.1 respectively) heres are the commands to reproduce.

Full commands in Windows (Powershell) + Python 3.7.7
```
mkdir test-pp
cd test-pp
python -m venv venv\
.\venv\scripts\activate
pip install --no-cache-dir pandapower==2.3.0 pytest
```
and then simply run the testsuite with
```
python
>>> import pandapower.test
>>> pandapower.test.run_all_tests()
```

lthurner, 2020-08-26:
> Though i don't get the error with pandapower 2.3.1 

If its fixed in 2.3.1 then all is well, no?

> It only show the following, no version print:
> <module 'packaging.version' from '/home/zhe/.local/lib/python3.6/site-packages/packaging/version.py'>

My bad, the correct command is

> pandapower.__version__

hitchenzhe, 2020-08-27:
> If its fixed in 2.3.1 then all is well, no?
>
>
> My bad, the correct command is

Thanks, it works now and I think this question could be closed. The reason is the python env.

closed 2020-09-01 by lthurner