e2nIEE/pandapower #874: A series of problem of not converging encountered in testing the tutorials

TravisCao, 2020-08-17:
Using the develop branch of pandapower, I copied the code in opf_basic.ipynb in the tutorials folder. However, the example network fails to converge and so with other example networks I tested. It seems that it is due to the version of pandapower since I saw the last editing time of opf_basic.ipynb is 14 months ago. Could you please send me a packed conda virtual environment using the version of pandapower and simbench at that time when you edited and tested the tutorials? My email is travisyjcao@gmail.com. Thanks a lot!

lthurner, 2020-08-19:
Yes the tutorial is not converging, its not a problem of the environment but with the opf solver. This issue has been open for a whil now, but we are looking into it now!

lthurner, 2020-08-26:
Fixed with #891

closed 2020-08-26 by lthurner