e2nIEE/pandapower #876: test_costs_pwl fails

miek770, 2020-08-17:
Hi,

I am currently following the instructions for the installation of the development version, as shown here: https://www.pandapower.org/start/#develop

I have a brand new virtual environment on Python 3.8.3 (Windows 10, 64 bits), and the tests failed because of missing dependencies (see issue X), and then because of this:

```
FAILED pandapower\test\opf\test_costs_pwl.py::test_get_costs - assert 0.10080294615700174 == (2 * array([0.05040147]))
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
============================================================== 1 failed, 403 passed, 5 skipped, 7 xfailed, 877 warnings in 264.95s (0:04:24) ==============================================================
```

Most likely a very easy fix, those values are quite close.

lthurner, 2020-08-17:
Not sure if this is actually the reason, but pandapower is not yet tested on Python 3.8

In any case, testing "==" to compare floats isn't best practice...

miek770, 2020-08-17:
Oops, installed Python 3.7.8, refollowed the steps (including #875), and got the same result:

```
$ python
Python 3.7.8 (tags/v3.7.8:4b47a5b6ba, Jun 28 2020, 08:53:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pandapower.test
>>> pandapower.test.run_all_tests()
=========================================================================================== test session starts ===========================================================================================
platform win32 -- Python 3.7.8, pytest-6.0.1, py-1.9.0, pluggy-0.13.1
rootdir: C:\Users\michela.lavoie\gits\pandapower\pandapower\pandapower\test, configfile: pytest.ini
collecting 159 items
No module named 'julia'
collected 537 items

...

FAILED pandapower\test\opf\test_costs_pwl.py::test_get_costs - assert 0.10080294615700174 == (2 * array([0.05040147]))
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
============================================================== 1 failed, 403 passed, 5 skipped, 7 xfailed, 877 warnings in 275.79s (0:04:35) ==============================================================
```

Using [pytest.approx](https://docs.pytest.org/en/stable/reference.html#pytest-approx) should solve this.

lthurner, 2020-08-18:
I think the issue is not the float comparison, but the fact that a float is compared to a single value array. I am assuming this is valid in some versions of numpy, but obviously not in the one you are using. Should be fixed now.

commit eca04b0a (2020-08-18): use np.isclose to compare floats Closes #876
commit 854ebf97 (2020-08-27): use np.isclose to compare floats Closes #876
closed 2020-08-18 by lthurner