e2nIEE/pandapower #877: Wrong state estimation estimated values

RebbahD, 2020-08-17:
Dear All, 

After using the measurements of the lines, for state estimation in addition to the bus, we face an issue. We are currently working with the 33 bus system, and not selecting all the buses, we would like to choose from bus 0 to 25, and in addition to that measurements of the lines, when we do that the estimated values for bus 0 which is the slack bus is supposed to be 1.0000 but it gives the value of : 0.97. Is this wrong? or this is how it is supposed to work?

We think that the slack bus is supposed to be 1.0000 always. isn't it?

0     0.970510
1     0.968301
10    0.939024
11    0.937924
12    0.934974
13    0.934276
14    0.933673
15    0.932939
16    0.932305
17    0.931963
18    0.967557
19    0.964249
2     0.959929
.
.
.
.

Best Regards, and looking forward to your reply as soon as possible. 

Dhiaa.

wangzhenassd, 2020-08-18:
No, slack in state estimation is only a degree reference the magnitude can vary.

wangzhenassd, 2020-08-18:
You can try reduce the std_dev of the vm at slack if you want this value to be close to 1.0 p.u.

lthurner, 2020-08-19:
Closed because the question was answered, feel free to reopen if you have more questions.

closed 2020-08-19 by lthurner