e2nIEE/pandapower #884: test_costs_pwl still fails

Hexelena, 2020-08-19:
In #876 the error in line 99 in `test_costs_pwl.py` was adressed. This was in function `test_get_costs()`. 
A similar error now also appears in the following function `test_cost_piecewise_linear_sgen()` in line 129. 

Just tested it and this worked for line 129 of test_costs_pwl.py
`assert np.isclose(net.res_cost, 2 * net.res_sgen.p_mw.values)`

not making a pull request since i am not sure what the difference between 
`net.res_sgen.p_mw.values` 
and
`net.res_sgen.p_mw.values[0]`
as fixed in line 99 is.

lthurner, 2020-08-19:
Thanks for reporting.

net.res_cost is a float, net.res_sgen.p_mw.values is an array. Can you please try if

> assert np.isclose(net.res_cost, 2 * net.res_sgen.p_mw.values[0])

also works for you? I can't reproduce the error unfortunately.

Hexelena, 2020-08-19:
Just tested it. This works too. 

(Just realised that it does since `net.res_sgen.p_mw.values` is a list consisting of a single float (since there's only one sgen present in the test net). Apparently np.isclose is willing to unpack that, but of course comparing the numbers themself is cleaner)

closed 2020-08-19 by lthurner