e2nIEE/pandapower #886: Convergence Error when Converting bus type (PV_to_PQ)

DRL2019, 2020-08-19:
Hi,

I want to convert PV bus to the PQ one for some networks such as IEEE14, IEEE24_rts, IEEE57, IEEE145, RTE2848, ... cases (by the attached short code: [PV2PQ.txt](https://github.com/e2nIEE/pandapower/files/5097834/PV2PQ.txt) )

Unfortunately, for those networks in above, with an exception about the IEEE14 case, I got the following convergence error:
pandapower.powerflow.LoadflowNotConverged: Power Flow nr did not converge after 10 iterations!

I increased the maximum iteration to 100, the problem did not resolved.
Does it mean there is a bug in pandapower for those networks (rather than IEEE14 case) or there is a problem in my code?
It is so strange that my code works ONLY for the IEEE14 case. Is there a big difference between the IEEE14 case and the other ones in the context?

I would be so glad if somebody, especially the pandapower developers, could leave a comment to solve this problem.

Best regards,

lthurner, 2020-08-24:
Can you please post a working minimal example that shows this problem?

FlorianShepherd, 2020-08-25:
You mix up all the indices of all elements. Consider your line

```
for i in net.gen.index.tolist():
	pp.create_sgen(net, bus=i, p_mw=abs(net.res_bus.p_mw[i] + net.load.p_mw[i]), \
							   q_mvar=(net.res_bus.q_mvar[i] + net.load.q_mvar[i]))
```

Here, you create an sgen at "some" bus, which has the bus index of the gen and add the p_mw value of the bus and load at that index.

What you probably want to do is to create an sgen at the bus of the generator:

```
for i in net.gen.index:
    bus = net.gen.at[i, "bus"]
    p_mw = abs(net.bus.at[i, "p_mw"] + net.load.loc[net.load.bus == i, "p_mw"])
    pp.create_sgen(net, bus=i, p_mw=p_mw)
```
or something like that.

DRL2019, 2020-08-26:
> Can you please post a working minimal example that shows this problem?
Dear @lthurner , 

Thank you for taking a look at my post.
First, let me explain a little bit more about what I want to do. I want to convert PV buses of a specific network to the PQ buses. So, I replace generators of the network with static generators.
Well, a simple example can be as follows:
"
import pandapower as pp
import pandapower.networks as nw
net = nw.case145()
net.switch.closed = True
net.gen.in_service = False
pp.runpp(net)
for i in net.gen.index.tolist():
	pp.create_sgen(net, bus=i, p_mw=abs(net.res_bus.p_mw[i] + net.load.p_mw[i]), \
						   q_mvar=(net.res_bus.q_mvar[i] + net.load.q_mvar[i]))
pp.runpp(net)
"

Now, by running this simple code, I got a convergence error at line 6 for this network and also other network such as case24_ieee_rts, case57, case2848rte. However, there is not any problem for case14. If this simple code is correct for converting the PV bus to the PQ bus in pandapower, I think that there is a problem in pandapower for those networks except for case14. 

Thank you for your reply.

Best regards

DRL2019, 2020-08-26:
> You mix up all the indices of all elements. Consider your line
> 
> ```
> for i in net.gen.index.tolist():
> 	pp.create_sgen(net, bus=i, p_mw=abs(net.res_bus.p_mw[i] + net.load.p_mw[i]), \
> 							   q_mvar=(net.res_bus.q_mvar[i] + net.load.q_mvar[i]))
> ```
> 
> Here, you create an sgen at "some" bus, which has the bus index of the gen and add the p_mw value of the bus and load at that index.
> 
> What you probably want to do is to create an sgen at the bus of the generator:
> 
> ```
> for i in net.gen.index:
>     bus = net.gen.at[i, "bus"]
>     p_mw = abs(net.res_bus.at[i, "p_mw"] + net.res_load.loc[net.load.bus == i, "p_mw"])
>     pp.create_sgen(net, bus=i, p_mw=p_mw)
> ```
> 
> or something like that.

Dear @FlorianShepherd ,

Thank you for your useful comment. Please let me divide my reply in two parts as follow:

1) As you told, I want to create sgen at "each" generator bus. 
But the error comes before indices, indeed, after the command " net.gen.in_service = False " by which I plan to make generators out of service. I mean, the error exactly takes place at the first time running power flow command: pp.runpp(net). 
One may ask me to put the command " net.gen.in_service = False " after creating sgen. There is no thing new with this, indeed, the error will be moved to the second time running power flow command.
 
It is strange for me that I do not have any problem with the current code when considering case14() as the network but there is convergence error for the other networks like case24_ieee_res, case57(), case145(), case2848rte().

2) About the mixing the indices, I applied your suggested commands, but I got a value error (see below) in that line in which we want to create sgen:
" ValueError: setting an array element with a sequence. "

Thank you for your reply in advance.

Best regards,

lthurner, 2020-08-26:
> Now, by running this simple code, I got a convergence error at line 6 for this network and also other network such as case24_ieee_rts, case57, case2848rte. However, there is not any problem for [...]

Yes, because you are deactivating the generators and then running a power flow without any generators. Not sure why you would expect that to work. 

This is not a problem with pandapower, your code isn't correct. We can support with pandapower questions but it seems to me you need to look up some power system and Python basics first. We unfortunately don't have time to debug your code...

DRL2019, 2020-08-27:
> Yes, because you are deactivating the generators and then running a power flow without any generators. Not sure why you would expect that to work.
>
> This is not a problem with pandapower, your code [...]

Dear @lthurner , 
Thanks for your reply. 
Since the code works for the case14 network, I expect that it should work for other networks in pandapower. Can you tell me why pandapower allows correctly running power flow for case14() by deactivating the generators and then running a power flow without any generators and does not allow to other networks?
I may not be a professional in Python or in Power System but I follow the pandapower documents:
" 
If you want to model a generator as PQ load with fixed reactive power and variable voltage, please use a static generator instead. ( https://pandapower.readthedocs.io/en/v2.3.1/elements/gen.html )
" 
When I find some problems in pandapower, I expect that one replies me in a normal way. For example, case14() in pandapower works even by deactivating the generators while other cases can't work. I'm not sure there are big differences between these networks in the context.

Best regards,

DRL2019, 2020-08-27:
> Yes, because you are deactivating the generators and then running a power flow without any generators. Not sure why you would expect that to work.
>
> This is not a problem with pandapower, your code [...]

Dear @lthurner , 
Thanks for your reply. 
Since the code works for the case14 network, I expect that it should work for the similar cases. I will be so happy if one could tell me why pandapower allows correctly running power flow for case14 even without any generator while it does not allow to other networks such as case24, case57, case145, case2848rte,...?
When I follow the pandapower documents to model generators as PQ loads:
" 
If you want to model a generator as PQ load with fixed reactive power and variable voltage, please use a static generator instead. ( https://pandapower.readthedocs.io/en/v2.3.1/elements/gen.html )
" 
and I find some problems in "pandapower", I expect that one replies me to resolve ...

Best regards,

DRL2019, 2020-08-27:
> You mix up all the indices of all elements. Consider your line
>
> ```
> for i in net.gen.index.tolist():
> 	pp.create_sgen(net, bus=i, p_mw=abs(net.res_bus.p_mw[i] + net.load.p_mw[i]), \ [...]

Dear @FlorianShepherd ,

Thank you for your useful comment.

As you told, I want to create sgen at "each" generator bus.
But the error comes before indices, indeed, after the command " net.gen.in_service = False " by which I plan to make generators out of service. I mean, the error exactly takes place at the first time running power flow command: pp.runpp(net).
One may ask me to put the command " net.gen.in_service = False " after creating sgen. There is no thing new with this, indeed, the error will be moved to the second time running power flow command.
Meanwhile, I can't keep generators to be in service while I want to replace them to static generators.

It is strange for me that there is not any convergence error when considering case14() without any generator as the network but there is convergence error for the other networks like case24_ieee_res, case57(), case145(), case2848rte() with making their generators out of service.
Do you think it is reasonable?

Thank you for your comment in advance.

Best regards,

lthurner, 2020-08-31:
> Since the code works for the case14 network, I expect that it should work for the similar cases. I will be so happy if one could tell me why pandapower allows correctly running power flow for case14 [...]

Some networks will converge without generators, because they are electrically stable even without the generators. Other networks are not electrically stable without the generators, because there is not enough reactive power and voltage stability in the network. This is not an issue with pandapower, its just comes down to what you are trying to calculate isn't physically feasible. 

>  and I find some problems in "pandapower", I expect that one replies me to resolve ...

You didn't find a problem in pandapower though, you are asking us to debug your code, which we don't have time for unfortunately.

closed 2020-08-31 by lthurner

DRL2019, 2020-08-31:
> Some networks will converge without generators, because they are electrically stable even without the generators. Other networks are not electrically stable without the generators, because there is [...]

Thanks for your comments. So, let me improve my question:
How can we convert PV buses to PQ buses in pandapower?
Based on what said in the pandapower documents, we need to use sgen instead of gen. In order to replace gen with sgen, if we need to make generators out of service, we got the convergence error in some networks. On the other hand, if we don't make generators of a network out of service, the network does not accept converting PV bus to PQ bus. 
Now, can you guide how we can convert a PV bus to a PQ bus in pandapower?

Best regards,

DRL2019, 2020-09-01:
Dear pandapower developers,

I solved the problem of pandapower for some small network such as case14 and case57, but there is still the convergence error for larger networks such as case145, case2848rte...

Best regards,

lthurner, 2020-09-01:
```
import pandapower as pp
import pandapower.networks as nw
net = nw.case2848rte()

pp.runpp(net)
for i in net.gen.index:
    bus = net.gen.bus.at[i]
    p_mw = net.res_gen.p_mw.at[i]
    q_mvar = net.res_gen.q_mvar.at[i]
    in_service = net.gen.in_service.at[i]
    pp.create_sgen(net, bus=bus, p_mw=p_mw, q_mvar=q_mvar, in_service=in_service)
    net.gen.in_service.at[i] = False
pp.runpp(net, init="results")
```

DRL2019, 2020-09-02:
Great! Thank you very much!
The main point was the including init="results" to runpp(net).

All the best,