e2nIEE/pandapower #887: Read data from excel for Estimation

maru159, 2020-08-20:
I have some problems with read data from excel for Estimation. I try read data as same as method in Youtube, i can't assign variable for bus and line. So that, my net.data have a problem with element. They can't know element in net. Some one can help me how to read data measure from excel for Estimation. Thanks for reading
[image]
[image]

lthurner, 2020-08-24:
I don't understand the problem, can you please post a reproducable minimal example?

lthurner, 2020-08-31:
Please reopen if you provide a better description or way to reproduce.

closed 2020-08-31 by lthurner