e2nIEE/pandapower #888: Branch currents for '1ph' short circuit

chefPony, 2020-08-20:
With release 2.3 `case='min'` for `fault='1ph'` is now supported. Does this include also calculation of the branch currents if option `branch_results=True` is specified?

lthurner, 2020-08-24:
It definitely wasn't actively tested, but I guess the branch calculation should work the same for all types of short-circuits. Did you check if it works or not?

chefPony, 2020-08-24:
I checked, it seem to be broken `res_line_sc`  is full of NaN even if `res_bus_sc` has values.
For example:
```
def three_bus_example():
    net = pp.create_empty_network()
    b1 = pp.create_bus(net, 110)
    b2 = pp.create_bus(net, 110)
    b3 = pp.create_bus(net, 110)

    pp.create_ext_grid(net, b1, s_sc_max_mva=100., s_sc_min_mva=80., rx_min=0.4, rx_max=0.4, x0x_max=0.2, x0x_min=0.1,
                      r0x0_max=0.3, r0x0_min=0.2)
    net.ext_grid['x0x_min'] =0.1
    net.ext_grid['r0x0_min'] =0.1
    pp.create_line(net, b1, b2, std_type="305-AL1/39-ST1A 110.0" , length_km=20.)
    pp.create_line(net, b2, b3, std_type="N2XS(FL)2Y 1x185 RM/35 64/110 kV" , length_km=15.)
    net.line['r0_ohm_per_km'] = 0.1
    net.line['x0_ohm_per_km'] = 0.1
    net.line['c0_nf_per_km'] = 0.1
    net.line["endtemp_degree"] = 80

    pp.create_sgen(net, b2, sn_mva=2, p_mw=0, k=1.2)
    return net

net = three_bus_example()

shortcircuit.calc_sc(net, fault='1ph', case='max', branch_results=True)

net.res_bus_sc

#   ikss_ka
#0	0.730547
#1	0.688562
#2	0.672599

net.res_line_sc

#   ikss_ka
#0	NaN
#1	NaN
```

wangzhenassd, 2021-01-04:
Should this be closed?

lthurner, 2021-01-12:
Yes can be closed, thanks a lot for pitching in @chefPony!

closed 2021-01-12 by lthurner