e2nIEE/pandapower #892: Problems in converting pandapower to PowerModels

gdgarcia, 2020-08-24:
Hi, guys.

I think I found some problems in the modules that convert the pandapower network to the PowerModels network.

Please, correct me if I'm wrong, but I think I found the following issues:

1. DClines: As reported in #838, the dclines' model is not ok for running an O.P.F.

2. Pandapower is building PowerModels data with loads, sgens, and storages connected to buses that are off, which pandapower does not load inside PowerModels.  Thus lead to an error inside PowerModels.

3. Pandapower is passing loads, sgen, storages, and gen elements in MW/MVAr, but PowerModels consider them in p.u.
    - Note: this problem can be hidden when baseMVA = 1.0 MVA (which is the case in some test networks in pandapower)
    - Note: I haven't checked yet if the same is happening in asymmetric_load and asymmetric_sgen

All these issues are causing convergence or inconsistent outputs problems.

As I'm working with pandapower and PowerModels, I'd like to create a pull_request to correct those issues, if I'm not missing something.

Thanks

gdgarcia, 2020-08-24:
>pandapower does have a connectivity check that works smoothly for the power flow. I think its deactivated for OPF by default, but in theory it should work just as well for OPF. A few adaptions might be necessary to make it work, but it should be possible. Please check this out as a first lead if you want to fix this.
 
 1. I solved this issue by just not writing to PowerModels elements connected to buses that are off. At this level, it's not a problem of checking the
connectivity of the elements, it's just about not loading elements connected to buses that do not exist in PowerModels
(because they're off) 
 
> I don't quite understand the difference, isn't MW/MVAr the same as pu?

2. They are not generally the same, except in case baseMVA == 1.0 MVA. As it's done today, pandapower pass to PowerModels loads, sgen, storage, and gen in MW/MVAr/MVA. This leads to the following: consider a bus K that has a load of PL[K] = 100 MW in a network where baseMVA = 200 MVA. PowerModels will receive 100 for PL[K] assuming the value is in p.u., which means that will be imposed on bus K a load of 100 p.u. (or 100 * 200 MW or 20 GW), when a load of just 100/200 or 0.5 p.u. should be imposed.

> You named gen elements, which are stored in the gen matrix, as well as loads/sgen/storages, which are stored in the bus matrix. Can you say which columns and values exactly this problems applies to?

3. In my analyses, I found that the problem lays in different places of pandapower:
3.1. For load, sgen and storage: They are treated inside pandapower.converter.powermodels.to_pm._pp_element_to_pm
All the elements that has "_mw" or "_mvar" should be devided by net.sn_mva
How it's now:
```python
            scaling = net[element].at[idx, "scaling"]
        if element == "sgen":
            pd = -net[element].at[idx, "p_mw"] * scaling
            qd = -net[element].at[idx, "q_mvar"] * scaling
        else:
            pd = net[element].at[idx, "p_mw"] * scaling
            qd = net[element].at[idx, "q_mvar"] * scaling
```
How should be:
```python
            scaling = net[element].at[idx, "scaling"]
        if element == "sgen":
            pd = -net[element].at[idx, "p_mw"] * scaling / net.sn_mva
            qd = -net[element].at[idx, "q_mvar"] * scaling  / net.sn_mva
        else:
            pd = net[element].at[idx, "p_mw"] * scaling  / net.sn_mva
            qd = net[element].at[idx, "q_mvar"] * scaling  / net.sn_mva
```
Also the following changes should be made inside ppc_to_pm bus loop:
```python
    for row in ppci["bus"]:
        bus = dict()
        idx = int(row[BUS_I]) + 1
        bus["index"] = idx
        bus["bus_i"] = idx
        bus["zone"] = int(row[ZONE])
        bus["bus_type"] = int(row[BUS_TYPE])
        bus["vmax"] = row[VMAX]
        bus["vmin"] = row[VMIN]
        bus["va"] = row[VA]
        bus["vm"] = row[VM]
        bus["base_kv"] = row[BASE_KV]

        # NOTE: Here the changes as made 
        # In powermodels, all loads should be in p.u. (flag per_unit = True)
        pd = row[PD] / baseMVA
        qd = row[QD] / baseMVA

```

The same applies to all other quantities that are expected to be in MW/MVAr/MVA in the power flow module but are expected to be in p.u. in PowerModels. For example the rows PG, QG, QMAX, QMIN, PMIN, and PMAX for generators.

Also, when these changes are made, some changes must be made in read_pm_results_to_net, to convert the p.u. values of PowerModels back to MW/MVAr/MVA of pandapower.

Please, correct me if I'm wrong. 
Thanks.

FlorianShepherd, 2020-08-25:
Thank you for looking into this @gdgarcia !

You're correct that the conversion is missing. However, I'm not a 100% sure at the moment if PowerModels.jl isn't doing this when calling the power flow. In the julia script, loading the converted pandapower grid, the command `correct_network_data` is called:

https://github.com/e2nIEE/pandapower/blob/dc5a2f6baa0c5d5173c881b43783748b8ed263f5/pandapower/opf/pp_2_pm.jl#L119

This function, in turn, calls `make_per_unit` inside the julia environment. 

https://github.com/lanl-ansi/PowerModels.jl/blob/f122169226d7b6d844964b3592f07d8ce8c9d58e/docs/src/network-data.md

Is that maybe already enough to solve the problem?

gdgarcia, 2020-08-25:
Hi, @FlorianShepherd.

Thank you for your reply.

The function correct_nettwork_data! really calls make_per_unit!. But take a look at its definition:

```julia
"Transforms network data into per-unit"
function make_per_unit!(data::Dict{String,<:Any})
    if !haskey(data, "per_unit") || data["per_unit"] == false
        data["per_unit"] = true
        if _IM.ismultinetwork(data)
            for (i,nw_data) in data["nw"]
                _make_per_unit!(nw_data)
            end
        else
            _make_per_unit!(data)
        end
    end
end
```

The flag "per_unit" is passed to Julia and its value is True, so the function correct_network_data! will actually call make_per_unit! but the changes won't be made.

Actually, the function make_per_unit! should NOT be called anyway, because it assumes that more than just the power quantities are not in p.u., and it'll change things that we don't want to. For instance, this function would reapply the convertion degree-to-radians to phase shift and max. and min. branch angle limits values.

Please, feel free to correct me if I'm missing something.

Thanks.

FlorianShepherd, 2020-08-25:
> Actually, the function make_per_unit! should NOT be called anyway on the data how it is now, because it assumes that more than just the power quantities are not in p.u., and it'll change things that we don't want to. For instance, this function would reapply the convertion degree-to-radians to phase shift and max. and min. branch angle limits values.

This means that the angle limits are all wrong when calling `correct_network_data!`? I never experienced any problems, probably because I had `baseMVA` set to 1. 

If I understand correctly, we have two options now:

1. Convert the data to the mixed format and call `correct_network_data!` -> Advantage: the data is more similar to the ppc / mpc from matpower, right?
2. Convert the data in p.u. and make sure that everything is correct for PowerModels.jl to run. -> Advantage: the format is correct and one exactly knows what PowerModels.jl is working with

The `correct_network_data!` function is quite nice, since it also does additional things, e.g., removing unnecessary cost terms within an OPF calculation. But if I see it correctly, then we could your suggestion (converting everything to p.u. values) and set the `["per_unit"]` flag to `true` inside the converted PM Dict. Then we can still call `correct_network_data!` and `_make_per_unit!` is skipped when called by it.

patripla, 2020-08-26:
> This means that the angle limits are all wrong when calling correct_network_data!? I never experienced any problems, probably because I had baseMVA set to 1.
Just adding to this: I had convergence issues using pypower if baseMVA was set to 1 (instead of 100) as some solver tolerances are scaled. If baseMVA was used as 100 for pypower and as 1 for powermodels, it seemed to work.

pplagowski, 2020-08-26:
> This means that the angle limits are all wrong when calling correct_network_data!? I never experienced any problems, probably because I had baseMVA set to 1.

Just adding to this: I had OPF convergence issues using pypower if baseMVA was set to 1 (instead of 100) as some solver tolerances are scaled. If baseMVA was used as 100 for pypower and as 1 for powermodels, it seemed to work.

gdgarcia, 2020-08-26:
Hi, folks.

@FlorianShepherd 

> This means that the angle limits are all wrong when calling `correct_network_data!`? I never experienced any problems, probably because I had `baseMVA` set to 1.

1. No. They are right, because a convertion to radians is applied before passing them to Julia. The problem would be applying the convertion twice (inside pandapower and inside _make_per_unit!_) since the angle limits would be made very small then.

> If I understand correctly, we have two options now:
> 1. Convert the data to the mixed format and call `correct_network_data!` -> Advantage: the data is more similar to the ppc / mpc from matpower, [...]

2. Correct, @FlorianShepherd. You're right to the point.

> The `correct_network_data!` function is quite nice, since it also does additional things, e.g., removing unnecessary cost terms within an OPF calculation. But if I see it correctly, then we could [...]

3. That's right. The benefits of _correct_network_data!_ can stay, regardless of which option is made.

Thanks!

lthurner, 2020-08-27:
>If I understand correctly, we have two options now:
>
>    Convert the data to the mixed format and call correct_network_data! -> Advantage: the data is more similar to the ppc / mpc >from matpower, right?
>    Convert the data in p.u. and make sure that everything is correct for PowerModels.jl to run. -> Advantage: the format is >correct and one exactly knows what PowerModels.jl is working with

I would heavily favor option 2 here. When we are building a data structure for PowerModels, we should aim to build the PowerModels data structure, and not a matpower or pypower data structure.

rbolgaryn, 2020-10-13:
Hi @gdgarcia @lthurner @FlorianShepherd @pplagowski 

what is the current state of this issue?

friederikemeier, 2020-12-02:
I would prefer putting everything to p.u. and with base_mva = 100 MVA. This is the Powermodels standard and should help us reduce issues of this kind. In the long run I need the fix to this myself, I can implement this, but am not sure when. @gdgarcia are you still on it? Otherwise I can take this from here.

friederikemeier, 2021-07-21:
This issue is kind of deprecated now that we introduce PandaModels.jl, I think I can close this for now.

closed 2021-07-21 by friederikemeier