e2nIEE/pandapower #893: state estimation

wjwhappy, 2020-08-24:
I want to set up a simple network with two nodes. Node 0 is connected to the external power grid, and node 1 is connected to a load. However, when I respectively calculated the power flow and estimated the state of this network, I found that the results of the two were quite different. I thought the power flow was correct while the state estimation was wrong, but I didn't know what the problem was.

here is the code:
`import pandapower as pp
from pandapower.estimation import estimate

net = pp.create_empty_network()

bus0 = pp.create_bus(net, vn_kv=10, type='n', name='BUS0')
bus1 = pp.create_bus(net, vn_kv=10, type='n', name='BUS0')

pp.create_ext_grid(net, bus0)  # set the slack bus to bus 1

pp.create_load(net, bus=bus1, p_mw=2, q_mvar=0.3)

line0 = pp.create_line(net, from_bus=bus0, to_bus=bus1, length_km=1, name="Line0-1",
                       std_type='NA2XS2Y 1x95 RM/25 12/20 kV')

pp.create_measurement(net, "v", "bus", 1.006, 0.01, element=bus0)  # V at bus 0
pp.create_measurement(net, "v", "bus", 0.906, 0.01, element=bus1)  # V at bus 1
pp.create_measurement(net, "p", "bus", -2.100, 0.01, element=bus0)  # P at bus 0
pp.create_measurement(net, "q", "bus", -0.286, 0.01, element=bus0)  # Q at bus 0

pp.runpp(net)
print(net.res_bus)

success = estimate(net, init='flat')
print(net.res_bus_est)`

here is the result:
`      vm_pu  va_degree      p_mw    q_mvar
0  1.000000   0.000000 -2.012969 -0.298729
1  0.993302  -0.098722  2.000000  0.300000
      vm_pu  va_degree      p_mw    q_mvar
0  0.952560   0.000000  2.099823  0.285926
1  0.959853   0.118215 -2.115309 -0.286252`

lthurner, 2020-08-24:
This looks like a problem with the signing system - @wangzhenassd can you please check?

wangzhenassd, 2020-08-24:
In state estimation, currently p, q are positive for injection and negative for consumption, which is also stated in documentation.
Maybe this could be changed to match the pandapower powerflow convention

```
value (float) - Measurement value. Units are “MW” for P, “MVar” for Q, “p.u.” for V, “kA” for I. Generation is a positive bus power injection, consumption negative
```

wjwhappy, 2020-08-24:
> In state estimation, currently p, q are positive for injection and negative for consumption, which is also stated in documentation.
> Maybe this could be changed to match the pandapower powerflow [...]

Thank you for your reply, for the generating bus, the injected power is positive and the consumption is negative, so for the bus 0, so i should change the create_measurement:

pp.create_measurement(net, "p", "bus", 2.100, 0.01, element=bus0)  # P at bus 0
pp.create_measurement(net, "q", "bus", 0.286, 0.01, element=bus0)  # Q at bus 0

and the result looks right:
      vm_pu  va_degree      p_mw    q_mvar
0  1.000000   0.000000 -2.012969 -0.298729
1  0.993302  -0.098722  2.000000  0.300000
      vm_pu  va_degree      p_mw    q_mvar
0  0.959449   0.000000 -2.100151 -0.286064
1  0.952202  -0.117092  2.084869  0.285819

lthurner, 2020-08-25:
> In state estimation, currently p, q are positive for injection and negative for consumption, which is also stated in documentation.
Maybe this could be changed to match the pandapower powerflow convention

Is there any reason to keep it that way? Because it is definitely confusing that it is different for power flow and state estimation.

wangzhenassd, 2020-08-25:
OK, i think it can be changed

closed 2020-08-31 by lthurner