e2nIEE/pandapower #894: issue on ploting and version

hitchenzhe, 2020-08-24:
Dear all

When I run:
`import pandapower as pp`

`print(pp.version)`

It shows <module 'packaging.version' from '/home/zhe/.local/lib/python3.6/site-packages/packaging/version.py'>
And the ploting of network doesn't return any figure.

I use pandapower 2.3.1. Other calculation functions work well.

Best

lthurner, 2020-08-27:
Try

> import matplotlib.pyplot as plt
> plt.show()

after plotting?

hitchenzhe, 2020-08-27:
> Try
> 
> > import matplotlib.pyplot as plt
> > plt.show()
> 
> after plotting?

Dear staff,
plt works. But another problem I meet is the plotly doesn.t work:
`simple_plotly(net)`
`pf_res_plotly(net, figsize=2, filename=dnet+'_b1.html')`

It shows following while return nothing
No or insufficient geodata available --> Creating artificial coordinates. This may take some time...

Best

lthurner, 2020-08-27:
Maybe you don't have all the relevant packages installed. You will need plotly as well as the pacakges necessary to create artificial coordinates: https://pandapower.readthedocs.io/en/v2.3.1/plotting/matplotlib/generic.html

hitchenzhe, 2020-08-27:
> Maybe you don't have all the relevant packages installed. You will need plotly as well as the pacakges necessary to create artificial coordinates: https://pandapower.readthedocs.io/en/v2.3.1/plotting/matplotlib/generic.html

Yes, I have installed python-igraph but no pycairo(maybe not necessary).

When I run pp.plotting.simple_plotly(net) individually, it returned:

`!pp.plotting.simple_plotly(net)

Figure({
    'data': [{'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [0.0, 0.5, 1.0],
              'y': [0.0, 0.0, 0.0]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [1.0, 1.5, 2.0],
              'y': [0.0, -0.375, -0.75]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [2.0, 2.5, 3.0],
              'y': [-0.75, -1.0, -1.25]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [3.0, 3.5, 4.0],
              'y': [-1.25, -1.25, -1.25]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [4.0, 4.5, 5.0],
              'y': [-1.25, -1.25, -1.25]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [5.0, 5.5, 6.0],
              'y': [-1.25, -1.5, -1.75]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [6.0, 6.5, 7.0],
              'y': [-1.75, -1.75, -1.75]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
              'x': [7.0, 7.5, 8.0],
              'y': [-1.75, -1.75, -1.75]},
             {'hoverinfo': 'text',
              'line': {'color': 'rgb(128, 128, 128)', 'width': 1},
              'mode': 'lines',
              'name': 'lines',
              'type': 'scatter',
[... 244 lines cut ...]
                             'Index: 28<br />Name: 29<br />V_n: 12.66 kV<br />Load: 0.120 MW<br />',
                             'Index: 29<br />Name: 30<br />V_n: 12.66 kV<br />Load: 0.200 MW<br />',
                             'Index: 30<br />Name: 31<br />V_n: 12.66 kV<br />Load: 0.150 MW<br />',
                             'Index: 31<br />Name: 32<br />V_n: 12.66 kV<br />Load: 0.210 MW<br />',
                             'Index: 32<br />Name: 33<br />V_n: 12.66 kV<br />Load: 0.060 MW<br />'],
                            dtype=object),
              'type': 'scatter',
              'x': [0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0, 10.0, 11.0,
                    12.0, 13.0, 14.0, 15.0, 16.0, 17.0, 2.0, 3.0, 4.0, 5.0, 3.0,
                    4.0, 5.0, 6.0, 7.0, 8.0, 9.0, 10.0, 11.0, 12.0, 13.0],
              'y': [0.0, 0.0, -0.75, -1.25, -1.25, -1.25, -1.75, -1.75, -1.75,
                    -1.75, -1.75, -1.75, -1.75, -1.75, -1.75, -1.75, -1.75, -1.75,
                    0.75, 0.75, 0.75, 0.75, -0.25, -0.25, -0.25, -0.75, -0.75,
                    -0.75, -0.75, -0.75, -0.75, -0.75, -0.75]}],
    'layout': {'autosize': True,
               'height': 102.94117647058823,
               'hovermode': 'closest',
               'margin': {'b': 5, 'l': 5, 'r': 5, 't': 5},
               'showlegend': True,
               'template': '...',
               'title': {'font': {'size': 16}},
               'width': 700.0,
               'xaxis': {'showgrid': False, 'showticklabels': False, 'zeroline': False},
               'yaxis': {'showgrid': False, 'showticklabels': False, 'zeroline': False}}
})`

rbolgaryn, 2020-12-02:
Hi @hitchenzhe ,

can you proviede a small example?

hitchenzhe, 2020-12-02:
> Hi @hitchenzhe ,
> 
> can you proviede a small example?

I just can not use plotly, I don know why, but matplot works

rbolgaryn, 2020-12-03:
Does this code work for you?

```python
import pandapower.networks as nw
from pandapower.plotting.plotly.simple_plotly import simple_plotly
from plotly.offline import plot
net = nw.mv_oberrhein()
fig = simple_plotly(net)
plot(fig)
```

closed 2021-02-17 by rbolgaryn

seilsepour, 2024-03-22:
Hi there
I have problem with simple_plotly too.
I can not run the above code. It resurns the following error:

```
WARNING:pandapower.auxiliary:numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
[<ipython-input-66-e049a360e961>](https://localhost:8080/#) in <cell line: 5>()
      3 from plotly.offline import plot
      4 net = nw.mv_oberrhein()
----> 5 fig = simple_plotly(net)
      6 plot(fig)

6 frames
[/usr/local/lib/python3.10/dist-packages/pandas/core/generic.py](https://localhost:8080/#) in blk_func(values)
   1556                 # [ExtensionArray, ndarray[Any, Any]]"; expected
   1557                 # "_SupportsPos[ndarray[Any, dtype[Any]]]"
-> 1558                 return operator.pos(values)  # type: ignore[arg-type]
   1559 
   1560         new_data = self._mgr.apply(blk_func)

TypeError: bad operand type for unary +: 'str'

```

seilsepour, 2024-03-22:
Hi @rbolgaryn,
can you help me? this is my code:
```
from pandapower.plotting.plotly import simple_plotly
simple_plotly(net) 
```
I got this error:
```
from pandapower.plotting.plotly import simple_plotly
net.sgen.scaling = 1
marker_trace = pplotly.create_weighted_marker_trace(net, "sgen", column_to_plot="p_mw", marker_scaling=200, color="green", patch_type="circle-open")
#simple_plotly(net, bus_size=0.25, additional_traces=[marker_trace])
simple_plotly(net)
```