e2nIEE/pandapower #897: Can't load network from json dictionary

chefPony, 2020-08-27:
I think it's a bug, the following example reproduce the behaviour:

```
net = pp.create_empty_network()
b1 = pp.create_bus(net, 110)
b2 = pp.create_bus(net, 110)
b3 = pp.create_bus(net, 110)

pp.create_ext_grid(net, b1, s_sc_max_mva=100., s_sc_min_mva=80., rx_min=0.4, rx_max=0.4, x0x_max=0.2, x0x_min=0.1,
                  r0x0_max=0.3, r0x0_min=0.2)
net.ext_grid['x0x_min'] =0.1
net.ext_grid['r0x0_min'] =0.1
pp.create_line(net, b1, b2, std_type="305-AL1/39-ST1A 110.0" , length_km=20.)
pp.create_line(net, b2, b3, std_type="N2XS(FL)2Y 1x185 RM/35 64/110 kV" , length_km=15.)
net.line['r0_ohm_per_km'] = 0.1
net.line['x0_ohm_per_km'] = 0.1
net.line['c0_nf_per_km'] = 0.1
net.line["endtemp_degree"] = 80

pp.create_sgen(net, b2, sn_mva=2, p_mw=0, k=1.2)

print(net)

#This pandapower network includes the following parameter tables:
#   - bus (3 elements)
#   - sgen (1 element)
#   - ext_grid (1 element)
#   - line (2 elements)

net_str = pp.to_json(net)

#Load net from json-string
net = pp.from_json_string(net_str)  

print(net)

#This pandapower network includes the following parameter tables:
#   - bus (3 elements)
#   - sgen (1 element)
#   - ext_grid (1 element)
#   - line (2 elements)

#OK

#Convert json-string to dict
net_dict = json.loads(net_str)

#Load net from json-dict
net_from_dict = pp.from_json_dict(net_dict)

net_from_dict 

print(net_from_dict)

#This pandapower network includes the following parameter tables:

# KO network is empty

```

lthurner, 2020-08-27:
There is a special Encoder/Decoder for pandapower networks that you need to use if you use json.loads directly:

> net_dict = json.loads(net_str, cls=pp.io_utils.PPJSONDecoder)

chefPony, 2020-08-28:
> There is a special Encoder/Decoder for pandapower networks that you need to use if you use json.loads directly:
> 
> > net_dict = json.loads(net_str, cls=pp.io_utils.PPJSONDecoder)

This would directly load the pandapowerNet from the string, it's not exactly what I mean. 
Is it not possible to load the pandapower from a json-dictionary instead of the json-string?

lthurner, 2020-08-28:
I don't understand what you are trying to do. I don't know if there is something like a json-dictionary?

chefPony, 2020-09-03:
Sorry, I explained poorly. I just mean dictionary, I am used to load json as dictionaries not string. I guess using strings it's fine but a bit clunky if your pandapowerNet is a section of another json where the other sections are dictionaries (as it is usually the case).
Instead of having:
```
json_input = {
                       "section_1": { ... },
                       "section_2": { ... },
                       "pandapowerNet_section: "pp_string"
                     }

net = json.loads(json_input["pandapowerNet_section"],  cls=pp.io_utils.PPJSONDecoder)
```

I would have preferred to have something like this:
```
json_input = {
                       "section_1": { ... },
                       "section_2": { ... },
                       "pandapowerNet_section: {...} #pandapowerNet dictionary
                     }

net = pp.net_from_dict( json_input["pandapowerNet_section"] )
```

rbolgaryn, 2020-10-13:
Hi @chefPony ,

The function pp_hook implements the functionality in the PPJSONDecoder, and it accepts a dictionary. 
Would it be helpful for your use case?

SteffenMeinecke, 2021-11-11:
Closed due to inactivity

closed 2021-11-11 by SteffenMeinecke