e2nIEE/pandapower #900: Storing pandapower networks - issues for developers

dlohmeier, 2020-08-28:
I would like to address a few aspects that I found recently when looking at the serialization of pandapower networks. Unfortunately the whole topic is quite tricky and gets even more complex with every release, but of course it is important to make the storing of pandapower networks possible. Here is what I have in mind:

1. The module io_utils is huge and will most probably not shrink, but grow further. Would it be possible to make a new package for serialization and add some modules inside to define different classes and functions so that developers will not have to check this huge file every time they try to change something?
2. I found that it is not possible to save two pandapower networks in one JSON file. Is that the desired behaviour? Find an example below.
3. What is the status of other io-operations? Is pickling possible and if it is not well suited, how do we address this?

There is a test on json encoding and decoding (test/api/test_file_io.py:test_json_encoding_decoding). Here, the following is possible:

    net = networks.mv_oberrhein()
    f = frozenset(['12', 3])
    d = {"a": net, "b": f}
    json_string = json.dumps([f, net, d], cls=PPJSONEncoder)
    f1, net1, d1 = json.loads(json_string, cls=PPJSONDecoder)

    assert f == f1
    assert d["b"] == d1["b"]
    assert pp.nets_equal(net, net1, exclude_elms=["line_geodata"])
    assert pp.nets_equal(d["a"], d1["a"], exclude_elms=["line_geodata"])

But the following will fail:

    net = networks.mv_oberrhein()
    small_net = networks.case5()
    f = frozenset(['12', 3])
    d = {"a": net, "b": f, "c": small_net}
    json_string = json.dumps([f, net, d, small_net], cls=PPJSONEncoder)
    f1, net1, d1, small_net1 = json.loads(json_string, cls=PPJSONDecoder)

    assert f == f1
    assert d["b"] == d1["b"]
    assert pp.nets_equal(net, net1, exclude_elms=["line_geodata"])
    assert pp.nets_equal(d["a"], d1["a"], exclude_elms=["line_geodata"])
    assert pp.nets_equal(small_net, small_net1 , exclude_elms=["line_geodata"])
    assert pp.nets_equal(d["c"], d1["c"], exclude_elms=["line_geodata"])

If you check what is returned by `json.loads`, you will find that net1 and small_net1 will both equal small_net.

dlohmeier, 2020-09-03:
I added a new PR #909 that addresses the second issue (without the workaround of saving text in the JSON file that has to be restored additionally). What do you think about the idea of making an io-package that contains different modules (for file_io, excel, json, (pickle) (de-)serialization) for more clarity?

SteffenMeinecke, 2021-11-11:
Since the PR has been closed: @dlohmeier Can this issue be closed?

dlohmeier, 2021-11-15:
It seems like the issues I described are not really urgent enough to be addressed by anyone currently. It's okay for me to close this issue.

closed 2021-11-15 by dlohmeier