e2nIEE/pandapower #908: TimeSeries  Simulation

dkothona, 2020-09-02:
Hello, 
Does anyone know if there is a possibility to keep at a variable, ex. a dataframe, the vm_pu values of the network at each timestep instead of using the OutputWriter?

rbolgaryn, 2020-10-13:
Hi @dkothona ,

we use the OutputWriter to store results for a variable. 
Why would you want to use a dataframe instead?
By the way, the output writer has a DataFrame with the results that you can access, would it help?
e.g. for vm_pu it would be net.output_writer.at[0, 'object']['output']['res_bus.vm_pu']

Roman

dkothona, 2020-11-30:
I use the results of the power flow the at an optimization problem. So I need to keep them in a dataframe at each time step. Finally, I did it but when you treat the problem as time series is more time consuming than run power flow in a for loop at each step. Thanks a lot for your answer!

dkothona, 2020-11-30:
> Hi @dkothona ,
>
> we use the OutputWriter to store results for a variable.
> Why would you want to use a dataframe instead?
> By the way, the output writer has a DataFrame with the results that you can [...]

closed 2020-11-30 by dkothona