e2nIEE/pandapower #910: maybe its a bug

wjwhappy, 2020-09-04:
I found that the PP_plot function affects pinv：
when i import pp_plot, it causes errors:
`
import numpy as np
from scipy.linalg import pinv
import pandapower.plotting as pp_plot

data = np.loadtxt('HPH.csv')
b = pinv(data)
print(b)
`
Traceback (most recent call last):
  File "E:/Document/Research/Paper/my paper/simulation/state_estimation/test.py", line 6, in <module>
    b = pinv(data)
  File "D:\Program\CodeLanguage\Python\Python37\lib\site-packages\scipy\linalg\basic.py", line 1311, in pinv
    x, resids, rank, s = lstsq(a, b, cond=cond, check_finite=False)
  File "D:\Program\CodeLanguage\Python\Python37\lib\site-packages\scipy\linalg\basic.py", line 1221, in lstsq
    raise LinAlgError("SVD did not converge in Linear Least Squares")
numpy.linalg.LinAlgError: SVD did not converge in Linear Least Squares

But if i delete this func, no problems:
`
import numpy as np
from scipy.linalg import pinv

data = np.loadtxt('HPH.csv')
b = pinv(data)
print(b)
`
[[ 2.84950147e-11  1.37075124e-10  6.46888723e-09  1.56583756e-10
   3.17290849e-09  5.10476578e-11  1.60147463e-09  2.84950147e-11
  -2.38473255e-10  8.59973656e-09 -1.56248256e-10  3.29338798e-09
   3.66262422e-11  1.58577853e-09  2.84950147e-11 -6.83772233e-11
   2.21026093e-09 -5.09763268e-11  1.64083794e-09  2.84950147e-11
  -1.02118610e-10  3.90747549e-13 -3.65988833e-11  1.60775520e-09]
 [ 1.37075124e-10  6.59399192e-10  3.11185493e-08  7.53245365e-10
   1.52632603e-08  2.45564500e-10  7.70388561e-09  1.37075124e-10
  -1.14717439e-09  4.13689893e-08 -7.51631445e-10  1.58428262e-08
   1.76190353e-10  7.62837961e-09  1.37075124e-10 -3.28928287e-10
   1.06324490e-08 -2.45221363e-10  7.89324261e-09  1.37075124e-10
  -4.91241057e-10  1.87968911e-12 -1.76058743e-10  7.73409828e-09]
 [ 6.46888723e-09  3.11185493e-08  1.46855520e-06  3.55473639e-08
   7.20308006e-07  1.15887479e-08  3.63563903e-07  6.46888723e-09
  -5.41377715e-08  1.95229680e-06 -3.54711993e-08  7.47659043e-07
   8.31482394e-09  3.60000604e-07  6.46888723e-09 -1.55228748e-08,...

here is the HPH.csv:
`
1.000000000000000000e+00 4.810494946034317820e+00 2.270182099151087982e+02 5.495128082732023245e+00 1.113496000081325406e+02 1.791459252773627497e+00 5.620192321949733127e+01 1.000000000000000000e+00 -8.368946562574741677e+00 3.017979337422399340e+02 -5.483354103673036661e+00 1.155776899686676984e+02 1.285356143811092977e+00 5.565108661756586628e+01 1.000000000000000000e+00 -2.399620563857219224e+00 7.756658330590204287e+01 -1.788955976121833658e+00 5.758333364501339702e+01 1.000000000000000000e+00 -3.583735999550128781e+00 1.371283898805586432e-02 -1.284396015446759520e+00 5.642233281618652541e+01
4.810494946034317820e+00 2.314086162582148276e+01 1.092069951454388729e+03 2.643428586979353767e+01 5.356466880820643155e+02 8.617805681493877046e+00 2.703590676048006003e+02 4.810494946034317820e+00 -4.025877514289710035e+01 1.451797434990644888e+03 -2.637764720303567501e+01 5.559858934685971690e+02 6.183199233657411753e+00 2.677092709151186796e+02 4.810494946034317820e+00 -1.154336259483507732e+01 3.731336569741916946e+02 -8.605763681812010901e+00 2.770043354751449556e+02 4.810494946034317820e+00 -1.723954391375720263e+01 6.596554264780925270e-02 -6.178580541013176664e+00 2.714193468557315327e+02
2.270182099151087982e+02 1.092069951454376223e+03 5.153726763306040084e+04 1.247494140596069883e+03 2.527838686860963935e+04 4.066938727005278906e+02 1.275886000307667018e+04 2.270182099151087982e+02 -1.899903267510912883e+03 6.851362667424191022e+04 -1.244821232946521377e+03 2.623824028281036226e+04 2.917992508713803090e+02 1.263381006375047218e+04 2.270182099151087982e+02 -5.447575648823442407e+02 1.760902689133704553e+04 -4.061255833161160353e+02 1.307246532503540038e+04 2.270182099151087982e+02 -8.135733314262076874e+02 3.113064159920764240e+00 -2.915812842488200545e+02 1.280889699516516885e+04
5.495128082732023245e+00 2.643428586979372241e+01 1.247494140596067837e+03 3.019643264562996521e+01 6.118803140056671737e+02 9.844298048986507510e+00 3.088367665870086967e+02 5.495128082732023245e+00 -4.598843327888820198e+01 1.658418301017480871e+03 -3.013173312265740833e+01 6.351142098841208963e+02 7.063196642168513506e+00 3.058098489067385231e+02 5.495128082732023245e+00 -1.318622234835296148e+01 4.262383102058352620e+02 -9.830542223158166948e+00 3.164277938100408960e+02 5.495128082732023245e+00 -1.969308833222572019e+01 7.535380661720345041e-02 -7.057920613830725109e+00 3.100479455514791880e+02
1.113496000081325406e+02 5.356466880820601091e+02 2.527838686860963571e+04 6.118803140056722896e+02 1.239873342197111015e+04 1.994782712272108256e+02 6.258061670178803979e+03 1.113496000081325406e+02 -9.318788522321315213e+02 3.360507920547929825e+04 -6.105692861469425452e+02 1.286952954787510134e+04 1.431238924813631002e+02 6.196726234883895813e+03 1.113496000081325406e+02 -2.671967899567907807e+02 8.637008025109684240e+03 -1.991995323733253542e+02 6.411881168507084112e+03 1.113496000081325406e+02 -3.990475700846536142e+02 1.526919136294986545e+00 -1.430169825720358006e+02 6.282604190608099998e+03
1.791459252773627497e+00 8.617805681493880599e+00 4.066938727005273790e+02 9.844298048986539484e+00 1.994782712272114225e+02 3.209326254348203289e+00 1.006834553752414649e+02 1.791459252773627497e+00 -1.499262675549258717e+01 5.406587008704979098e+02 -9.823205445259274526e+00 2.070527221085713734e+02 2.302663156939833033e+00 9.969665404794497476e+01 1.791459252773627497e+00 -4.298822462267855471e+00 1.389573733693946167e+02 -3.204841736228138060e+00 1.031581958639102226e+02 1.791459252773627497e+00 -6.420117015892064849e+00 2.456599228693925191e-02 -2.300943126097654279e+00 1.010783101866304605e+02
5.620192321949733127e+01 2.703590676047913348e+02 1.275886000307667200e+04 3.088367665870105725e+02 6.258061670178804889e+03 1.006834553752418913e+02 3.158656173570273040e+03 5.620192321949733127e+01 -4.703508921379023491e+02 1.696162429998430889e+04 -3.081750463199487626e+02 6.495688457505929364e+03 7.223948730418112518e+01 3.127698097162032354e+03 5.620192321949733127e+01 -1.348632906858310321e+02 4.359391159357050128e+03 -1.005427664130601784e+02 3.236294096239742430e+03 5.620192321949733127e+01 -2.014128554856652329e+02 7.706879239270157944e-01 -7.218552624356564706e+01 3.171043616800240670e+03
1.000000000000000000e+00 4.810494946034317820e+00 2.270182099151087982e+02 5.495128082732023245e+00 1.113496000081325406e+02 1.791459252773627497e+00 5.620192321949733127e+01 1.000000000000000000e+00 -8.368946562574741677e+00 3.017979337422399340e+02 -5.483354103673036661e+00 1.155776899686676984e+02 1.285356143811092977e+00 5.565108661756586628e+01 1.000000000000000000e+00 -2.399620563857219224e+00 7.756658330590204287e+01 -1.788955976121833658e+00 5.758333364501339702e+01 1.000000000000000000e+00 -3.583735999550128781e+00 1.371283898805586432e-02 -1.284396015446759520e+00 5.642233281618652541e+01
-8.368946562574741677e+00 -4.025877514289688008e+01 -1.899903267510921069e+03 -4.598843327888788224e+01 -9.318788522321331129e+02 -1.499262675549258539e+01 -4.703508921379016670e+02 -8.368946562574741677e+00 7.003926656723170652e+01 -2.525730780184278046e+03 4.588989747731449853e+01 -9.672635111736109366e+02 -1.075707688143207363e+01 -4.657409700516270163e+02 -8.368946562574741677e+00 2.008229626937651702e+01 -6.491505907285962849e+02 1.497167696696221029e+01 -4.819118421700295016e+02 -8.368946562574741677e+00 2.999209507461033652e+01 -1.147620167121443796e-01 1.074904161845789829e+01 -4.721954882744723250e+02
3.017979337422399340e+02 1.451797434990612146e+03 6.851362667424192477e+04 1.658418301017484737e+03 3.360507920547929825e+04 5.406587008704954087e+02 1.696162429998430889e+04 3.017979337422399340e+02 -2.525730780184285777e+03 9.108199281108542345e+04 -1.654864938465552996e+03 3.488110801924511907e+04 3.879178283250863046e+02 1.679538295169180128e+04 3.017979337422399340e+02 -7.242005279374967586e+02 2.340943456916655850e+04 -5.399032171494069416e+02 1.737853111205505775e+04 3.017979337422399340e+02 -1.081564119741912009e+03 4.138506472335977016e+00 -3.876280635686000551e+02 1.702814346084207864e+04
-5.483354103673036661e+00 -2.637764720303560750e+01 -1.244821232946518649e+03 -3.013173312265757886e+01 -6.105692861469447053e+02 -9.823205445259180379e+00 -3.081750463199475689e+02 -5.483354103673036661e+00 4.588989747731485380e+01 -1.654864938465554133e+03 3.006717222626797437e+01 -6.337534005827438932e+02 -7.048062885847948600e+00 -3.051546141782934569e+02 -5.48335410367 [... cut ...]
5.642233281618652541e+01 2.714193468557274969e+02 1.280889699516516339e+04 3.100479455514804386e+02 6.282604190608099998e+03 1.010783101866287694e+02 3.171043616800238851e+03 5.642233281618652541e+01 -4.721954882744726660e+02 1.702814346084206773e+04 -3.093836301864438951e+02 6.521162889538189120e+03 7.252279213344041864e+01 3.139964130718726210e+03 5.642233281618652541e+01 -1.353921900865159103e+02 4.376487578700063750e+03 -1.009370694782507911e+02 3.248986015584458073e+03 5.642233281618652541e+01 -2.022027452919676307e+02 7.737103652382507635e-01 -7.246861945132081928e+01 3.183479640420519900e+03

`

jhmenke, 2020-09-21:
What happens if you import in a different order?

```
import pandapower.plotting as pp_plot
import numpy as np
from scipy.linalg import pinv
```

wjwhappy, 2020-09-22:
> What happens if you import in a different order?
> 
> ```
> import pandapower.plotting as pp_plot
> import numpy as np
> from scipy.linalg import pinv
> ```

it shows the same problem:
Traceback (most recent call last):
  File "E:/Document/Research/Paper/my paper/simulation/state_estimation/test.py", line 6, in <module>
    b = pinv(data)
  File "D:\Program\CodeLanguage\Python\Python37\lib\site-packages\scipy\linalg\basic.py", line 1311, in pinv
    x, resids, rank, s = lstsq(a, b, cond=cond, check_finite=False)
  File "D:\Program\CodeLanguage\Python\Python37\lib\site-packages\scipy\linalg\basic.py", line 1221, in lstsq
    raise LinAlgError("SVD did not converge in Linear Least Squares")
numpy.linalg.LinAlgError: SVD did not converge in Linear Least Squares

rbolgaryn, 2020-11-11:
Hi @wjwhappy ,

I cannot reproduce the error, can you try updating your packages?
can you provide a smaller example for the data?

rbolgaryn, 2020-12-02:
Hi @wjwhappy , closing this for now, cannot reproduce. If you can provide a consise example, please reopen.

Roman

closed 2020-12-02 by rbolgaryn