e2nIEE/pandapower #913: Strange behaviors in re/active power deviations in some large networks!?

DRL2019, 2020-09-05:
Hi,
I'm studying on re/active power deviations in some small and large IEEE networks through pandapower. I get the following steps:

- After converting all PV buses to PQ buses in a small (large) network such as case57 (case145), I run the power flow. 
- Then I make a perturbation on the re/active power of an arbitrary sgen bus, e.g. bus 46 (122), of the case57 (case145) network and run again the power flow. 
- Finally, I depict plots of re/active power deviations for all buses, i.e, 
Delta_P or Delta_Q vs. bus numbers with 
Delta_P = p_mw_after - p_mw_before, 
where p_mw_after(before) denotes the active power after (before) imposing perturbation. Similarly, for Delta_Q.

[image]

[image]

[image]

[image]

As you can see in plots, in the larger network, the re/active power deviations of many buses are affected by imposing just 1% perturbation in the re/active power of a sgen bus (e.g. bus 122 in the case145). However, in the smaller network (case57), by imposing 10% (or even more) perturbation in the re/active power of a sgen bus (which sgen for case57 is created at bus 46) does not affect the re/active power deviations of other buses.
One can check and see these strange behaviors in the other large networks in pandapower.

Is it reasonable (acceptable) or there is a problem in the network which should be resolved?

Many thanks for your helpful comments in the context in advance.

Best regards,

wangzhenassd, 2020-11-16:
Hi, can you share the code to recreate the network and this behaviour?

Houshyar20, 2020-12-07:
Dear @wangzhenassd,

Sorry for the delay.
I attached my code here
[StrangeBehavior_PQ.txt](https://github.com/e2nIEE/pandapower/files/5653787/StrangeBehavior_PQ.txt)

As you can see, I considered both cases (with and without converting the PV bus to PQ bus) and you can check strange behaviours of variations of  active & reactive powers.

I will be so happy if you could leave your comments to resolve the problem (if there is any), or at least to understand it.

Best regards,

wangzhenassd, 2020-12-08:
In case145, there are also many shunts, since they are passive elements with constant Z, the P, Q of the shunts is related to the voltage on bus, which can be affected by perturb P, Q on other buses.

Please reopen on further questions

closed 2020-12-08 by wangzhenassd