e2nIEE/pandapower #923: State Estimation error for most of the available simbench datasets

shan-rak, 2020-09-13:
Hi,

I'm new to pandapower and had few questions about the state estimation module.

I tried accessing the simbench dataset and estimating. But for almost all modules i receive measurement capacity error
I understand from the documentation the min available measuremnts for estimation should be 2n-k elements.
Does this mean I should be adding dummy elements to the simbench dataset.

the "1-MVLV-semiurb-4.201-2-no_sw" has 39 measurement elements. But the algo expects 341 elements.

Can you let me know whats the solution around other than creating so many measurements.

Eg code:
sb_gridcode = "1-MVLV-semiurb-4.201-2-no_sw"
net = sb.get_simbench_net(sb_gridcode)
print(net)

from pandapower.estimation import estimate
success = estimate(net, init='flat')
print(success)

Error:
System is not observable (cancelling)
Measurements available: 16. Measurements required: 341
ERROR:root:An unexpected error occurred while tokenizing input

wangzhenassd, 2020-09-17:
Hi, MV and LV network has low measurement density as it is in the reality, if you want the WLS state estimation to work you need to manually add pseudo measurement

shan-rak, 2020-09-19:
Hi,
Thanks for your reply.
But my observation is if I add many pseudo measurements with value 0 and std deviation of (1/0.1/0.001) the matrix is linear.
So does that mean for MV and LV nw the WLS estimator cannot be used or am I doing something wrong.

wangzhenassd, 2020-09-19:
You are right, by adding many pseudo measurements with unrealistic values will cause convergence problem with WLS. A first try would be using the power flow value with small random noises to see if WLS converges.
WLS is by itself more suitable for meshed HV and EHV state estimation with high measurement redundancy.

wangzhenassd, 2020-09-22:
Please reopen on further questions

closed 2020-09-22 by wangzhenassd