e2nIEE/pandapower #925: Possibility to use custom solvers in OPF

hoppjo, 2020-09-14:
Hi,

would you consider integrating a generic way to replace the build-in optimizer used by the optimal power flow (OPF) with custom functions?
On the one hand, this would provide the possibility to change the optimization Problem before solving it (for example adding additional variables and constrains). On the other hand, this would open up the possibility to develop custom interfaces to third party optimization tools like Pyomo (see https://github.com/e2nIEE/pandapower/issues/699#issuecomment-600456745).

I currently use the following "hacky" solution:

1. call 'runopp()' / 'rundcopp()' but pickle the optimization problem instead of solving it.
2. modify and solve the problem externally and pickle the solution.
3. rerun 'runopp()' / 'rundcopp()' but read the solution form the pickle-file instead of calling PIPS

For this propose I changed pandapower as shown is this patch:

```patch
diff --git a/pandapower/pypower/dcopf_solver.py b/pandapower/pypower/dcopf_solver.py
index b5b2bc1f..239b1f4e 100644
--- a/pandapower/pypower/dcopf_solver.py
+++ b/pandapower/pypower/dcopf_solver.py
@@ -238,6 +238,8 @@ def dcopf_solver(om, ppopt, out_opt=None):
 #    else:
 #        raise ValueError("Unrecognised solver [%d]." % alg)
 
+    opt['om'] = om
+
     ##-----  run opf  -----
     x, f, info, output, lmbda = \
             qps_pypower(HH, CC, A, l, u, xmin, xmax, x0, opt)
diff --git a/pandapower/pypower/pips.py b/pandapower/pypower/pips.py
index a9e6a1b7..833d7609 100644
--- a/pandapower/pypower/pips.py
+++ b/pandapower/pypower/pips.py
@@ -18,12 +18,54 @@ from pandapower.pypower.pipsver import pipsver
 from scipy.sparse import vstack, hstack, eye, csr_matrix as sparse
 from scipy.sparse.linalg import spsolve
 
+import numpy as np
+from scipy.sparse import isspmatrix
+import dill
+from pathlib import Path
+import os
 
 EPS = finfo(float).eps
 
 
-def pips(f_fcn, x0=None, A=None, l=None, u=None, xmin=None, xmax=None,
-         gh_fcn=None, hess_fcn=None, opt=None):
+def hash_args(*args):
+    args_hash = 0
+    for arg in args:
+        if isinstance(arg, np.ndarray):
+            # print(arg, hash(arg.tobytes()))
+            args_hash += hash(arg.tobytes())
+        elif callable(arg):
+            try:
+                y = arg(args[1])
+            except TypeError:
+                pass
+            args_hash += hash_args(*y)
+        elif isspmatrix(arg):
+            args_hash += hash(str(arg.data))
+    # print(args, args_hash)
+    return hash(args_hash)
+
+
+class NotYetCalculated(Exception):
+    pass
+
+
+def pips(*args, **kargs):
+    if not os.environ.get('EMS_RD_PICKLE_AC_OPT_PROBLEM', '0') == '1':
+        return orig_pips(*args, **kargs)
+    args_hash = hash_args(*args)
+    result_file = Path(f'{args_hash}_result.p')
+    if result_file.is_file():
+        with open(f'{args_hash}_result.p', 'rb') as pfile:
+            result = dill.load(pfile)
+            return result
+    else:
+        with open(f'{args_hash}_call.p', 'wb') as pfile:
+            dill.dump((args, kargs), pfile)
+        raise NotYetCalculated(f'not yet calculated. hash: {args_hash}', args_hash)
+
+
+def orig_pips(f_fcn, x0=None, A=None, l=None, u=None, xmin=None, xmax=None,
+              gh_fcn=None, hess_fcn=None, opt=None):
     """Primal-dual interior point method for NLP (nonlinear programming).
     Minimize a function F(X) beginning from a starting point M{x0}, subject to
     optional linear and nonlinear constraints and variable bounds::
diff --git a/pandapower/pypower/pipsopf_solver.py b/pandapower/pypower/pipsopf_solver.py
index d8bc9a87..74dcb8e4 100644
--- a/pandapower/pypower/pipsopf_solver.py
+++ b/pandapower/pypower/pipsopf_solver.py
@@ -92,7 +92,8 @@ def pipsopf_solver(om, ppopt, out_opt=None):
              'max_red': max_red,
              'step_control': step_control,
              'cost_mult': 1e-4,
-             'verbose': verbose  }
+             'verbose': verbose,
+             'om': om}
 
     ## unpack data
     ppc = om.get_ppc()
diff --git a/pandapower/pypower/qps_pypower.py b/pandapower/pypower/qps_pypower.py
index 6f3d9230..e63d2501 100644
--- a/pandapower/pypower/qps_pypower.py
+++ b/pandapower/pypower/qps_pypower.py
@@ -15,9 +15,46 @@ from pandapower.pypower.qps_pips import qps_pips
 
 from pandapower.pypower.util import have_fcn
 
+import numpy as np
+from scipy.sparse import isspmatrix
+import dill
+from pathlib import Path
+import os
+
+
+def hash_args(*args):
+    args_hash = 0
+    for arg in args:
+        if isinstance(arg, np.ndarray):
+            #print(arg, hash(arg.tobytes()))
+            args_hash += hash(arg.tobytes())
+        elif isspmatrix(arg):
+            args_hash += hash(str(arg.data))
+    # print(args, args_hash)
+    return args_hash
+
+
+class NotYetCalculated(Exception):
+    pass
+
+
+def qps_pypower(*args, **kargs):
+    if not os.environ.get('EMS_RD_PICKLE_DC_OPT_PROBLEM', '0') == '1':
+        return orig_qps_pypower(*args, **kargs)
+    args_hash = hash_args(*args)
+    result_file = Path(f'{args_hash}_result.p')
+    if result_file.is_file():
+        with open(f'{args_hash}_result.p', 'rb') as pfile:
+            result = dill.load(pfile)
+            return result
+    else:
+        with open(f'{args_hash}_call.p', 'wb') as pfile:
+            dill.dump((args, kargs), pfile)
+        raise NotYetCalculated(f'not yet calculated. hash: {args_hash}', args_hash)
+
 
-def qps_pypower(H, c=None, A=None, l=None, u=None, xmin=None, xmax=None,
-                x0=None, opt=None):
+def orig_qps_pypower(H, c=None, A=None, l=None, u=None, xmin=None, xmax=None,
+                     x0=None, opt=None):
     """Quadratic Program Solver for PYPOWER.
 
     A common wrapper function for various QP solvers.

```

friederikemeier, 2020-10-14:
That's a good idea!
Though I have a question: Why use pickle? For all other applications we use json, e.g. for the export to the powermodels optimization. So I would prefer to have it in a json file.
And another suggestion: Instead of changing the runopp() I would prefer to add a new function run_custom_opf(), that can share most of it's code with runopp but then does your "hacky trick". And then of course it would be great if you provided a test for this as well ;)
I suggest a new function name, because it may be confusing, as the OPF is already configurable in so many ways, that this would overkill it.

rbolgaryn, 2020-12-02:
has not been updated for a while. @hoppjo if you want to update or discuss further, feel free to reopen.

closed 2020-12-02 by rbolgaryn