e2nIEE/pandapower #926: Combination of controllable but out-of-service and not controllable but in-service generators lead to IndexError

hoppjo, 2020-09-14:
If a network contains a controllable but out-of-service and a  not controllable but in-service generator, an `IndexError` is raised during a DC optimal power flow.

Minimal Example:
```python
import pandapower as pp

net = pp.create_empty_network()

bus = pp.create_bus(net, vn_kv=110., name="Bus 2")

pp.create_ext_grid(net, bus=bus, vm_pu=1.00, name="Grid Connection")

pp.create_gen(net, bus=bus, p_mw=5, name='gen 1')
pp.create_gen(net, bus=bus, p_mw=5, name='gen 3')

net.gen[['in_service', 'controllable']] = [[False, True],
                                           [True, False]]

pp.rundcopp(net)

``` 

Output:
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-3-3e64703d4458> in <module>
----> 1 pp.rundcopp(net)

/data/ssd/johannesh/projekte/pandapower/pandapower/run.py in rundcopp(net, verbose, check_connectivity, suppress_warnings, switch_rx_ratio, delta, trafo3w_losses, **kwargs)
    402     _check_bus_index_and_print_warning_if_high(net)
    403     _check_gen_index_and_print_warning_if_high(net)
--> 404     _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
    405 
    406 

/data/ssd/johannesh/projekte/pandapower/pandapower/optimal_powerflow.py in _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
     39     init_results(net, "opf")
     40 
---> 41     ppc, ppci = _pd2ppc(net)
     42 
     43     if not ac:

/data/ssd/johannesh/projekte/pandapower/pandapower/pd2ppc.py in _pd2ppc(net, sequence)
    113         aux._set_isolated_buses_out_of_service(net, ppc)
    114 
--> 115     _build_gen_ppc(net, ppc)
    116 
    117     if "pf" in mode:

/data/ssd/johannesh/projekte/pandapower/pandapower/build_gen.py in _build_gen_ppc(net, ppc)
     51     _init_ppc_gen(net, ppc, f)
     52     for element, (f, t) in gen_order.items():
---> 53         add_element_to_gen(net, ppc, element, f, t)
     54     net._gen_order = gen_order
     55 

/data/ssd/johannesh/projekte/pandapower/pandapower/build_gen.py in add_element_to_gen(net, ppc, element, f, t)
     80         _build_pp_ext_grid(net, ppc, f, t)
     81     elif element == "gen":
---> 82         _build_pp_gen(net, ppc, f, t)
     83     elif element == "sgen_controllable":
     84         _build_pp_pq_element(net, ppc, "sgen", f, t)

/data/ssd/johannesh/projekte/pandapower/pandapower/build_gen.py in _build_pp_gen(net, ppc, f, t)
    202         ppc = _check_gen_vm_limits(net, ppc, gen_buses, gen_is)
    203         if "controllable" in net.gen.columns:
--> 204             ppc = _enforce_controllable_vm_pu_p_mw(net, ppc, gen_is, f, t)
    205 
    206 

/data/ssd/johannesh/projekte/pandapower/pandapower/build_gen.py in _enforce_controllable_vm_pu_p_mw(net, ppc, gen_is, f, t)
    166     # if there are some non controllable gens -> set vm_pu and p_mw fixed
    167     if np.any(not_controllable):
--> 168         bus = net["gen"]["bus"].values[not_controllable]
    169         vm_pu = net["gen"]["vm_pu"].values[not_controllable]
    170         p_mw = net["gen"]["p_mw"].values[not_controllable]

IndexError: boolean index did not match indexed array along dimension 0; dimension is 3 but corresponding boolean dimension is 2

```

Would you consider this to be a bug. In that case I would provide a pull request.

friederikemeier, 2020-11-23:
Yeah, this should only be executed for the intersection of eg_is and not_controllable. But I don't have the time to fix this this month.

Crueda96, 2021-04-16:
Hello @friederikemeier, sorry for the inconvenience, there is currently a solution for this problem

friederikemeier, 2021-07-21:
This is related to #1178  where the same bug is reported.