e2nIEE/pandapower #929: DC net modeling

fdellekart, 2020-09-17:
Hello!

Is it possible to model parts of the network with direct current and connect it to parts of the network running on alternating current with an inverter.

I haven't seen an element that models an inverter, also I am not to sure if I understood the use of the DC Line element correctly.

If this is not supported, would it be desirable for pandapower to support connection of DC and AC networks in the future?

I just dived into pandapower a little bit, so please do not hesitate to clearify any misconseptions about its functionality from my side ;)

Thank you for your answers

lthurner, 2020-09-18:
> If this is not supported, would it be desirable for pandapower to support connection of DC and AC networks in the future?

You understood everything correctly. Its not supported yet, but would be very desirable! We have it on our todo list, but its not clear when we will get around to it. Contributions are always welcome!

fdellekart, 2020-09-18:
Thanks for your answer!

Would be an interesting project, but I don't think I am able to do it yet.

I'am in the last year of my bachelor "Energie-, Mobilitäts- and Umweltmanagement" at FH Joanneum in Kapfenberg, Austria. We have to do two bachelor thesis this year, I already have a topic for the second one but I am still thinking about what should be the topic for the first (smaller) thesis.

What would you think about the idea to write about pandapower to get some basic overview about how it works? Then I could keep it in the back of my mind to maybe get onto some project like the inverter in my master.

Could I contact you somewhen in the course of this semester if I choose pandapower as a topic and have questions?

rbolgaryn, 2020-11-06:
Hi @fdellekart ,

Thank you for the suggestion!
Feel free to contribute a pull request when you have time.

Roman

closed 2024-08-11 by fdellekart