e2nIEE/pandapower #937: Error while saving to excel

Davere, 2020-10-01:
Hi,
recently updated to the newest pp Version and now I have problems saving to excel with geodata.

The Error code is:
  File "C:/Users/name/Documents/scripte/multi_layer_opf/real_grid_opf.py", line 679, in <module>
    pp.to_excel(net, "C:/Users/drhl86/Documents/OPF/try" + ".xlsx")
  File "C:\Users\name\AppData\Local\Continuum\anaconda3\lib\site-packages\pandapower\file_io.py", line 70, in to_excel
    include_empty_tables=include_empty_tables)
  File "C:\Users\name\AppData\Local\Continuum\anaconda3\lib\site-packages\pandapower\io_utils.py", line 116, in to_dict_of_dfs
    geo = coords_to_df(value, geotype="bus")
  File "C:\Users\name\AppData\Local\Continuum\anaconda3\lib\site-packages\pandapower\io_utils.py", line 63, in coords_to_df
    k = max(len(v) for v in value.coords.values)
  File "C:\Users\name\AppData\Local\Continuum\anaconda3\lib\site-packages\pandapower\io_utils.py", line 63, in <genexpr>
    k = max(len(v) for v in value.coords.values)
TypeError: object of type 'NoneType' has no len()

Process finished with exit code 1

The problem comes from buses, which have no coords. The default value is None and so line 63 in pandapower\io_utils.py is trying to get len(None).
I got it working for me with adding an if statement and setting the len for not list elements to 0:
k = max(len(v) if isinstance(v,list) else 0 for v in value.coords.values) 

I also used a newly created grid to test this and altough the default parameter is None for coords is the result for buses without this parameter coords = nan, which results in an error in pandapower\io_utils.py line 69, where only None is continued in the for loop.

Here the grid creation example:
import pandapower as pp
    net = pp.create_empty_network("test")
    coords = list([(1,2),(4,5)])
    pp.create_bus(net,0.4, index = 1, geodata=[1,2], coords=[coords])
    pp.create_bus(net,0.4, index = 2, geodata=[1,7])
    pp.create_bus(net,0.4, index = 3, geodata=[1,7])
    pp.create_bus(net,0.4, index = 4, geodata=[1,7])

pawellytaev, 2024-11-26:
sounds that this is obsolete now with the current geojson developments? @KS-HTK @vogt31337

vogt31337, 2024-11-26:
@pawellytaev probably. WE should have a Look into the Tests, but i think saving to Excel ist covered.

KS-HTK, 2024-11-27:
For the given example this will raise a UserWarning as coords is not supported for bus any more.
It would be a good idea to change this for if geodata and coords are set. And just logging a warning instead of raising.

Removing the coords this works fine for me. The geojson string is dumped into the excel. It also is loaded back without any apperent issue.

There are test for from_excel and to_excel.

closed 2024-11-27 by KS-HTK