e2nIEE/pandapower #941: DepreciationWarning: ipragh --> jgraph error in /pandapower/plotting/generic_geodata.py

JoachimSchaeffer, 2020-10-06:
DeprecationWarning: To avoid name collision with the igraph project, this visualization library has been renamed to 'jgraph'. Please upgrade when convenient.

Should be easy to fix. Let me know whether there are any questions. 

[image]

JoachimSchaeffer, 2020-10-06:
P.S.: Easy fix is pip uninstall igraph. 
(Of course you cannot use igraph functions anymore)
The error occured in the plotting_basic notebook from the powerpandas repo.

rbolgaryn, 2020-11-06:
Hi @JoachimSchaeffer ,

is the errior still occurring with the latest version of igraph?

rbolgaryn, 2020-12-01:
cannot reproduce, haven't gotten any updates, closing the issue

closed 2020-12-01 by rbolgaryn