e2nIEE/pandapower #943: inconsistend consideration of unloaded branches

SteffenMeinecke, 2020-10-08:
If lines are connected to buses which are out of service, runpp behaves as there is a open switch at the end of the line so that the unloaded line behaviour is considered. Due to the splitting of trafo3ws into 3 trafo2ws, this behaviour is also given for trafo3ws.
In contrast, trafo2ws and impedances, which are connected to out of service buses, are not considered anymore.
It should be discussed whether this is a desired behaviour and if not who can change that.

Here an exemplifying code snippet:

```
import pandapower as pp

net = pp.create_empty_network()
pp.create_buses(net, 9, [110] + [110]*2 + [20]*2 + [20, 10] + [20]*2, in_service=[
    True] + [True, False]*4)

pp.create_ext_grid(net, 0)

pp.create_line(net, 0, 1, 1.1, "122-AL1/20-ST1A 110.0")
pp.create_switch(net, 1, 0, "l", closed=False)
pp.create_line(net, 0, 2, 1.1, "122-AL1/20-ST1A 110.0")

pp.create_transformer(net, 0, 3, '25 MVA 110/20 kV')
pp.create_switch(net, 3, 0, "t", closed=False)
pp.create_transformer(net, 0, 4, '25 MVA 110/20 kV')

pp.create_transformer3w(net, 0, 5, 6, '63/25/38 MVA 110/20/10 kV')

pp.create_impedance(net, 0, 7, 1e-4, 1e-3, net.sn_mva)
pp.create_impedance(net, 0, 8, 1e-4, 1e-3, net.sn_mva)

pp.runpp(net)
print(net.res_line[["p_from_mw", "q_from_mvar", "loading_percent"]])
print(net.res_trafo[["p_hv_mw", "q_hv_mvar", "loading_percent"]])
print(net.res_trafo3w[["p_hv_mw", "q_hv_mvar", "loading_percent"]])
print(net.res_impedance[["p_from_mw", "q_from_mvar"]])
```

rbolgaryn, 2020-11-06:
Hi @lthurner ,

can you take a look?

Roman

lthurner, 2020-11-11:
The reason this is implemented that way is the following:

1. Lines: it is more accurate to consider the open loop line loading capacity for the lines. This approach was adopted from commercial software.
2. 2W-Trafos: It used to be implemented in the same way than lines, with transformers being in open loop operation when connected to an OOS bus. However, in some cases there where convergence issues due to this bus on the out of service side of the transformer when there was a tap changer. The NR had difficulties with this one bus that had a very different voltage than the neighbouring buses and didn't converge. To avoid this problem, the solution was switched to setting the transformer out of service in this case. The open loop losses didn't seem important enough to risk non-convergence of the power flow.
3. 3W-Trafos: Behaviour follows from 2W trafo behaviour.
4. Impedance: impedances are an equivalent element that represent multiple element. Since an impedance is not a phyisical element like a line, it doesn't make any sense for it to be in "open loop" operation. Therefore, it is set out of service.

SteffenMeinecke, 2020-11-11:
Ok, that is reasonable. Then, I suggest, we can close this issue and have documented the behavior and its reason.

closed 2020-11-11 by SteffenMeinecke